nordicsemi/Android-BLE-Library #1: Necessity to compile for VERSION_1_8

peabody124, 2018-01-11:
It might be worth adding to the instructions to use this, the need for adding

+    compileOptions {
+        targetCompatibility JavaVersion.VERSION_1_8
+        sourceCompatibility JavaVersion.VERSION_1_8
+    }

to the gradle files for all the modules that include this. I was getting a fairly cryptic error
code '''com.android.dx.cf.code.SimException: invalid opcode ba (invokedynamic requires --min-sdk-version >= 26)''' 
that I eventually tracked down to this. Alternatively, I'm not sure how necessary the lambda 
expression in BleManager is since without that this restriction might not be required.

philips77, 2018-01-11:
True. Is it a problem for you using Java 1.8, Android Studio 3+, etc? I could revert it back to be 1.7 compatible, but 1.8 is there from some time, so I assumed it won't be much problem already.

peabody124, 2018-01-11:
No, it wasn't much of a problem other than the cryptic error.

philips77, 2018-01-31:
So let's close the issue now. If anyone finds this a blocker, please open a new issue and I'll revert it back to 1.7 (although I woudn't like to... lambdas are cool :) ).

closed 2018-01-31 by philips77

stanleyguevara, 2019-11-28:
I think it's still worth mentioning somewhere in the readme.

I've bumped into it on latest Android Studio (3.5.2 on Win10) with fresh project straight from the wizard (though AS install is pretty old, maybe it has default somewhere to not use 1.8, IDK).

Simply adding the library to gradle.build was causing quite cryptic crash:
`ClassNotFoundException: Didn't find class "com.test.MainActivity" on path: DexPathList`

Took a while to discover it was the lib in first place, as it wasn't crashing on API 28, only few commits later when running on API 26 it started crashing...

philips77, 2019-11-28:
Whaaaaat? No one reads readme?! I'll rename it to Pleasereadme or Youdbetterreadme ;)

PS. Ok, I'll put it higher in the TL;DR section.