nordicsemi/Android-BLE-Library #3: How to send more than 20 bytes of data

LuckyStar521, 2018-01-31:
I want to send more than 20 bytes in the data, how to deal with.

philips77, 2018-01-31:
Hi,
So there are 3 options how you can achieve it:
1. You may request higher MTU from the phone using [this method](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L809). MTU (Maximum Transfer Unit) is the size of the packet that can be sent using one write operation (it may be split into multiple packets on lower layer if DLE (Data Length Extension) is not uses, but that's transparent for the application). Highest MTU supported by Android (Lollipop and newer only) is 517 (you always deduct 3 bytes from MTU to get the max packet size), but it actually depends on the target device whether it supports higher MTUs, or not, and what is the max value. Usually max MTU is around 180-250 if they are supported by peripherals. What you can do is you can request 517 and you'll get the max supported value using [this callback](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1382). The default MTU is 23 - that means you may send 20 bytes per packet.
2. If you are using Write With Response (write type WRITE_TYPE_DEFAULT in Android), and your device supports Long Write (that means it reserves more bytes in a buffer for each write), you may just write longer packet despite having shorter MTU. Android will automatically chunk data into at-most-MTU-size packets and will sent it to the peripheral. For you it will be transparent. This, however, does not work with Write Without Response (WRITE_TYPE_NO_RESPONSE).
3. If your device does not support higher MTUs, or you are using Write Without Response, or Long Write is not possible, you have to split your data into at most 20-bytes chunks. Have a look at [this example](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/uart/UARTManager.java#L139). In nRF Toolbox UART sample, when the data are to be written using Write Without Response, they are chunked. After each chunk is sent the buffer window is moved MTU-bytes right and next part is sent (see [here](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/uart/UARTManager.java#L105)).

Right now the BLE library does not support sending longer packets automatically, but I guess it should, so I'll add this functionality soon. I also plan to add some more related features (callbacks, etc).

philips77, 2018-02-20:
I'll add support for this feature soon.

philips77, 2018-06-19:
This should now work with version 2 of the library.

closed 2018-06-19 by philips77

sreechary, 2018-06-27:
how can I send more that 20 bytes over BLE in swift 3 . I am using write without response .

philips77, 2018-06-27:
Hmm... It's an Android library, not iOS. But to answer your question, your device needs to support longer mtu. iOS will automatically request high mtu immediately after connecting. Use CBPeripheral.maximumWriteValueLength(for:) to get the maximum supported length.

Ne1c, 2018-08-16:
Hi, I send 2 packages but after the first package, flag mOperationInProgress became true and when I send the second package, package doesn't sending because mOperationInProgress == true. How I can fix it?

philips77, 2018-08-16:
Hi, please open a new issue. Describe what version of the library are you using.

balazsnemeth, 2018-09-12:
@philips77 , even with the latest vestion of the lib, the MTU exchange is not supported for Android 4 (like KitKat), right? 

Here is a exchange flow of [Nordic SoftDevice S132](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.api.v6.0.0%2Fgroup___b_l_e___g_a_t_t_c___m_t_u___e_x_c_h_a_n_g_e.html) , the `requestMtu` does implement this? Do you know any way to support large MTU for devices running Android 4.4 or lower (basically for SDK 19 and 20)?

philips77, 2018-09-13:
Can't be done. Support for higher mtu was added in Lollipop.

vamshik113, 2019-01-04:
> Hi,
> So there are 3 options how you can achieve it:
>
> 1. You may request higher MTU from the phone using [this [...]

Hi,
As you mentioned if DLE is not in use the long packet will be broken down in fragments in link layer. Is there a way to enable LE DLE in android or is there any phones that come with DLE enabled? Peripheral is however requesting for LL_Length_Req and android is responding with Unknown Request.

philips77, 2019-01-07:
No, this can't be enabled on Android. DLE support was added in Android 6, as far as I remember, and most newer phones do use it. Phones updated to 6 may not support this feature. There is also no API to check if DLE is working or now, other than just trying to calculate the bandwidth and estimating based on that.