nordicsemi/Android-BLE-Library #4: Integrate with the Bond Management Service

eliotstock, 2018-02-20:
Great to see all this code in a separate repo! I'll be migrating to it ASAP.

One feature request though: it would be great to see some integration with the bond management service so that I can request the deletion of the current bond with deleteBond() or some such method.

philips77, 2018-02-20:
Thanks for the suggestion. I'll look into it.
Right now I'm working on improving sending/receiving long data (automatic splitting) and adding per-request callbacks.

philips77, 2018-05-05:
Hi, please check out the develop branch. Also, there is BLE Common Library (also under development, develop branch) where i could add Bond Management Service.

philips77, 2018-10-02:
HI @eliotstock, could you give some more input on this task?

eliotstock, 2018-10-04:
Afraid I haven't been able to integrate with the develop branch yet - we need something pretty stable for now. I'll update this issue if I can find the time. Feel free to close it in the meantime if you like.

eliotstock, 2018-11-30:
Closing this for now, since it looks like there's nothing special about the bond management service that would make sense to integrate in the BLE library. I'll reopen if I hit any issues when using the service as I would any other service.

closed 2018-11-30 by eliotstock