nordicsemi/Android-BLE-Library #6: Data lost during sync using notifications

svvashishtha, 2018-03-17:
Hi,
I am using this library to connect to a sensor that's using characteristic notification to send data. 
I enable the notifications and I am able to receive data but the data is getting lost somewhere between being received by Android device and being received in Gattcallbacks.
The sensor also uses a nordic processing unit and updates notification characteristic every 20ms.
No matter the size of data, BlinkyManager reports data size to be approx 8000 notification updates.

However, the data size can be up to 10 Mb and I can only receive less than 1 Mb.

philips77, 2018-03-17:
Hi, could you write what's exactly the issue? I don't follow...

svvashishtha, 2018-03-17:
Hi,
The issue is that I'm only receiving part of the data that the device receives. The logs on sensor peripheral indicate that data is being read by the device but GattCallback is not updating the app.

svvashishtha, 2018-03-19:
Hi, I ran some more tests, while the Log output for my app is free of any errors, I get this when "No Filters" option is selected for Logcat.
`? E/JavaBinder: !!! FAILED BINDER TRANSACTION !!!  (parcel size = 160)
? E/BtGatt.JNI: An exception was thrown by callback 'btgattc_notify_cb'.
? E/Watchdog: !@Sync 1475 [2018-03-19 11:56:36.577]
? E/BtGatt.JNI: android.os.DeadObjectException: Transaction failed on small parcel; remote process probably died
                    at android.os.BinderProxy.transactNative(Native Method)
                    at android.os.BinderProxy.transact(Binder.java:761)
                    at android.bluetooth.IBluetoothGattCallback$Stub$Proxy.onNotify(IBluetoothGattCallback.java:406)
                    at com.android.bluetooth.gatt.GattService.onNotify(GattService.java:1322)`

and Since this error is repeated a lot of times, I'm guessing this happens for every data point lost.

philips77, 2018-03-19:
Are you sure that the ble manager is alive? Hasn't been killed with activity, service, garbaged, etc?
Do you have the instance to it in the service?

svvashishtha, 2018-03-19:
Yes, I even plotted the data that is received in the app. And like I said, not all of the data is missing.
But some data points in between, I was able to verify that the last and first timestamp were there in the data.

That would not be possible if BleManager had died. I'm using an Activity to manage Ble connection.

svvashishtha, 2018-03-20:
Hi, just wondering if you can do a speed test and include it in the Readme so that there's no confusion regarding this matter.
Things to test:

1. Maximum data throughput on various devices.

2. Minimum interval in the notification updates for reliable data transfer between Android and Ble peripheral.

This would set the performance standard for any app using this library.

bnchrnck, 2018-10-31:
Hi guys,

We have been experiencing a similar problem to the one outlined above. We have our android devices connected to a sensor via BLE, and the data being collected is the transferred to the mobile device and plotted in a graph on our app. However, in some cases and some mobile devices the data is incomplete or missing. We know this because we can see a complete data set on one device and another displays breaks/gaps in the generated graph.

Any thoughts as to why this might be? How should we go about diagnosing the problem?

philips77, 2018-10-31:
My idea is that in here: https://github.com/NordicSemiconductor/Android-BLE-Library/blob/develop/ble/src/main/java/no/nordicsemi/android/ble/MainThreadBluetoothGattCallback.java#L135, when the *safe* method is called from the UI thread, the value inside the characteristic might have been already changed. I need to pass the data as byte array. That is of course if you are using v2 of the lib.

bnchrnck, 2018-11-30:
Hi @philips77, it's possible I misunderstood your previous comment, do you require our data as byte array in order to debug this issue? Or is this something you can test independently? Thanks for your time.

philips77, 2018-11-30:
Hi, in the 2.0.0 the reference to value received in `onCharacteristicChanged` is copied as quickly as possible and then not used, unless you use deprecated `onCharacteristicNotified(BluetoothGatt, BluetoothGattCharacteristic)` method. If you are using the new callbacks you should get the proper data every time. Are you saying this doesn't work for you and you still get overwritten data in the callbacks?

bnchrnck, 2018-12-03:
Hi @philips77, thanks for the information and your efforts, we will test with 2.0.3 and get back to you with our results.

philips77, 2018-12-17:
Any success?

philips77, 2019-01-31:
The issue has been fixed in 2.0.3. If you are using BLE Common Library please wait until this https://github.com/NordicSemiconductor/Android-BLE-Common-Library/issues/2 task is resolved.

closed 2019-01-31 by philips77