nordicsemi/Android-BLE-Library #10: Refactor BleManager to get rid of generic part?

tamberg, 2018-05-04:
Looking at [class BleManager<E extends BleManagerCallbacks>] (https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L77) I wonder if the generic part <...> is more than a historic left-over? It seems E is only ever used to keep a reference to [mCallbacks](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/8f401a40ad8795c15135429f4670c9fe6e2800b5/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L96). The field itself seems to only be accessed as type BleManagerCallbacks. Maybe it's possible to replace the <...> part with a simple set method to set the callback of type BleManagerCallbacks?

tamberg, 2018-05-04:
In a way, the setter mentioned above is already there: https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L420 (just change the type E to BleManagerCallbacks)

philips77, 2018-05-04:
It really well integrates later on when you set the callbacks in the service/activity. Also, the mCallbacks available from the overriding manager has correct type, do you may directly use it there, without casting or keeping a second reference with another type. Check let's say csc sample in the nrf Toolbox.

tamberg, 2018-05-04:
Ok, I see. Thanks for the quick reply.

closed 2018-05-05 by philips77