nordicsemi/Android-BLE-Library #9: Refactor mCallbacks to become a method?

tamberg, 2018-05-04:
When implementing a Manager it seems to be required to access mCallbacks (https://github.com/NordicSemiconductor/Android-BLE-Library/blob/develop/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L134). This direct access can be somewhat confusing (to me at least, it is, every time I come across it). How about converting the field to a method, to make it more explicit that it's a part of the inherited interface?

philips77, 2018-05-04:
Good catch. The mCallbacks is required, but now it must be set before connecting which isn't obvious. I'll fix it. Thanks!

philips77, 2018-05-05:
I'll leave it as is now, just adding some debug logs if connect() is called without callbacks set. This is because some profiles in nRF Toolbox use static manager and they set the activity as callbacks each time it was recreated. This is bad design, but very simple and easy to understand for beginners. And it would again break compatibility, which is broken already...

closed 2018-08-13 by philips77