nordicsemi/Android-BLE-Library #11: Connection with Nordic UART device not working on Android 7.1

fabianenye, 2018-05-21:
I have a Motorola G5S with Android 7.1. The Nordic NRF Toolbox app and the NRF Connect and also this library, disconnects automatically inmediately after it connects. I cannot establish a connection. In Android 7.0 it works. Does it make sense?

sivapraka, 2018-07-19:
I have also faced the same issue

philips77, 2018-10-02:
Hi, sorry for the super long delay. Do you still face this issue? Did you try v2 of the BLE library?

hyoungtaikkim, 2018-11-12:
Hi,

In my case, Android 8.0 doesn't work for the matter of connection. I think that, in BLE manager,

mBluetoothGatt = device.connectGatt(mContext, false, mGattCallback = getGattCallback());

should be changed to 

mBluetoothGatt = device.connectGatt(mContext, false, mGattCallback = getGattCallback(), 2); 

especially for newer OS.

philips77, 2018-11-13:
The new version of the library uses TRANSPORT_LE for Marshmallow or newer: https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L805. Actually, the `connectGatt` with `transport` parameter was available before (since KitKat?), but only using reflections.

hyoungtaikkim, 2018-11-15:
Hi,

Thank you very much for your work again. Anyway I have tried the library 2.0, but there was some connection issue again for Android 8.0, LG phone. So I have gone back to 1.2 version with the above change I have commented in the above. So no connection problem in my case after the change. So I would like to recommend you to test the library 2.0 with Android 8.0 (Oreo).

Thanks again.

closed 2018-11-19 by philips77