nordicsemi/Android-BLE-Library #12: Suggestion - Retry the operation that triggered bonding but failed

weliem, 2018-06-11:
Reading an encrypted characteristic will trigger the bonding process to start. However, the read/write that triggered it sometimes fails. I see this happening a lot on Android 6 and far less on 7 & 8. When it happens I get a GATT_AUTH_FAIL (error 137) and I re-try the operation which then always succeeds.

It seems that on later Android versions (7 & 8), the stack retries the operation itself. I don't have proof for it other than I don't see any read/write failed errors on those Android versions.

So my suggestion would be to add an extra check and to retry the operation that triggered the bonding process. That way the users of the BleManager don't have to worry about it. Here is what I do:
```
public void onCharacteristicWrite(BluetoothGatt gatt, BluetoothGattCharacteristic characteristic, int status) {
            // Perform some checks on the status field
            if(status!= GATT_SUCCESS) {
                if (status == GATT_AUTH_FAIL) {
                    // Characteristic encrypted and needs bonding, so retry operation
                    // This only seems to happen on Android 6
                    HBLogger.v(TAG, "Write failed (Authentication failed), bonding in progress so retrying write command");
                    retryCommand();
                    return;
                } else if (status == GATT_INSUFFICIENT_AUTHENTICATION) {
                    // Characteristic encrypted and needs bonding, so retry operation
                    // This only seems to happen on Android < 6
                    HBLogger.v(TAG, "Write failed (Insufficient authentication), bonding in progress so retrying write command");
                    retryCommand();
                    return;
                }
...
```
Same thing is done in onCharacteristicRead of course....

philips77, 2018-06-11:
Thanks! I'll have a look at it tomorrow.
I'm developing v2 of the lib on develop branch. But I'll include your suggestion in both versions.

weliem, 2018-06-11:
Great! Will have a look once it out!

closed 2018-06-11 by weliem