nordicsemi/Android-BLE-Library #13: Question about delay when calling discoverServices for bonded devices

weliem, 2018-06-11:
I saw you recently added a delay in onConnectionStateChange before calling discoverServices for bonded devices. I also observed the need for this during testing with several devices so this is definately needed in some cases!

I have a couple of questions:

- I only have a limited set of devices and phones at my disposal and I only observed this issue on Android 6. On Android 7 & 8 the delay doesn't seem necessary. Your code doesn't look at the Android version but I think this is only needed for Android < 7. Did you also observe the need for a delay on Android 7 & 8?

- You use a 1600 ms delay. The device I encountered that needed a delay was the A&D UA-651BLE and with that one 600 ms seemed enough. Why are you suggesting 1600 ms? Seems quite long...

- What is your theory about why the delay is needed? My guess is that Android is doing the key exchange with the bonded peripheral and service discovery cannot be started until this has finished. I don't know enough about low-level Bluetooth to be sure...

philips77, 2018-06-11:
The main reason is that onConnectionStateChanged is/was called just after establishing the connection, before resuming security and before optional Service Changed indication could be received. When calling discoverServices() in onConnectionStateChanged, it would return cached services. When calling it just after SC indication is received, it would cancel the internal service discovery initiated be the indication and start it again. When called after the internal discovery ended, it immediately returns services.
The sd duration depends on number of services and connection interval. 1600ms is kind a long, but with nice ui unnoticeable. Guess it depends on phones and applications as well...

weliem, 2018-06-11:
Ok, but perhaps that is then a different scenario. In my case I had to add a delay because otherwise onServicesDiscovered would never be called. I am also not using the SC indication but I am calling refresh() after a device disconnects, so I forced the full service discovery to take place every time a connection is established.

weliem, 2018-06-11:
You say correctly that the service discovery depends on the number of services. But the 1600 ms delay in your code is BEFORE starting the discovery! So what is Android doing in that timeframe? Definitely not discovering services....my guess is that is still exchanging keys....for lack of a more plausible explanation...in my case it can't be the Service Changed indication because I am not doing any firmware updating

weliem, 2018-06-11:
Nonetheless, I will check if the A&D BPM supports this Service Changed indication.....perhaps that is causing the delay....although I am not touching it

philips77, 2018-06-11:
If you'd check sniffer, it does service discovery, actually. An internal one, for some unknown purpose. You'll never getting onServicesDiscovered after it. But when you'll call discoverServices while it discovers, it starts again. If after it's done, you get the result immediately, so there's no extra delay here.

weliem, 2018-06-11:
Indeed....it has the Service Changed characteristic. But my code is not using it....perhaps Android does something with it anyway....

philips77, 2018-06-11:
At least that's what I observed during testing on some phones.

weliem, 2018-06-11:
Thanks, you are probably right. It must be doing discovery anyway. I'll do some more testing this week to confirm. It would be really a pain if this delay would be only for devices with the SC characteristic since I never know this upfront....But since it only happens when you are already bonded I could check it at the initial pairing flow and then store this information so I would know if the delay is needed when reconnecting or not...so it becomes on a per-device basis.

closed 2018-08-13 by philips77