nordicsemi/Android-BLE-Library #17: Device disconnects automatically inmediately after connection

AvantStudioSL, 2018-07-05:
Hi all,

I  am facing some connection problems with the library. I downloaded it 4 days ago and  yesterday I started calling the BLEManager _connect(device)_ method. All devices are correctly scanned but when I start the connection, I can connect to the device but immediately it disconnects from the device. I have no clue about what can be causing this problem. Here is the trace I get from logcat:

`D/TEST-LP: onDeviceConnecting: LUIS-05B6
 D/BluetoothGatt: connect() - device: DB:AB:2A:56:B6:05, auto: false
    registerApp()
 D/BluetoothGatt: registerApp() - UUID=282d31c8-d3d0-4969-88a7-b233d4c2babc
 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=DB:AB:2A:56:B6:05
 D/TEST-LP: onDeviceConnected: LUIS-05B6
 D/TEST-LP: WearableDeviceListener in MAIN ACTIVIT
 D/BluetoothGatt: discoverServices() - device: DB:AB:2A:56:B6:05
 D/BluetoothGatt: onConnectionUpdated() - Device=DB:AB:2A:56:B6:05 interval=6 latency=0 timeout=500 status=0
 D/BluetoothGatt: onSearchComplete() = Device=DB:AB:2A:56:B6:05 Status=0
 D/BluetoothGatt: cancelOpen() - device: DB:AB:2A:56:B6:05
 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=DB:AB:2A:56:B6:05
 D/TEST-LP: onDeviceDisconnected: LUIS-05B6`

If you want to see some part of the code just let me know. Any help is apreciated!

AvantStudioSL, 2018-07-05:
Hi, thanks for the quick reply! I have been able to initialize the device now. What I was doing wrong was in the discover services. I have commented all the code inside "isRequiredServiceSupported" in the GattCallback and now it works, I have to rework this part of the code but at least I am not blocked. 

Let you know if I have other issue with the library!

philips77, 2018-07-05:
Still I recommend switching to v2 of the lib :)
Better earlier then later, I think. I'd love to get any feedback.

AvantStudioSL, 2018-07-05:
Hi again, one question: how is managed the library in terms of processes? I am having problems when performing UI operations in the callback methods (update screen in onDeviceConnected), I have tried with runOnUiThread but I still get errors. Do you have any idea about how should I manage these UI events?

Thanks in advance

philips77, 2018-07-05:
You'll find an example in nRF Toolbox app, available on GitHub. Updates are coming via local broadcast receiver, so they are called on UI thread. There should be no problems.

AvantStudioSL, 2018-07-05:
Ok, I will check that code, thank you very much!

AvantStudioSL, 2018-07-13:
Hi again!

I'm facing a strange problem when connecting to BLE devices, sometimes 'onConnectionUpdated' is not called when connecting to it. This is the trace I am getting for a correct connection:
` D/BluetoothGatt: connect() - device: D7:B8:25:14:13:9B, auto: false registerApp()
 D/BluetoothGatt: registerApp() - UUID=13b2e0fc-5363-4e38-b030-bc1aeb9926a2
 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=D7:B8:25:14:13:9B
 D/BluetoothGatt: onConnectionUpdated() - Device=D7:B8:25:14:13:9B interval=6 latency=0 timeout=500 status=0
 D/BluetoothGatt: onConnectionUpdated() - Device=D7:B8:25:14:13:9B interval=36 latency=0 timeout=500 status=0
 D/BluetoothGatt: discoverServices() - device: D7:B8:25:14:13:9B
 D/BluetoothGatt: onSearchComplete() = Device=D7:B8:25:14:13:9B Status=0
 D/NetworkSecurityConfig: No Network Security Config specified, using platform default
 D/TEST-LP: onBatteryValueReceived: 100
 D/BluetoothGatt: setCharacteristicNotification() - uuid: d44bc439-abfd-45a2-b575-925416129601 enable: true
 D/Companion API: deviceConnected(): connected to wearable device MK101-9B13[D7:B8:25:14:13:9B]`

But after this successful connection right after I disconnect and connect again this is the trace I get:
` D/BluetoothGatt: connect() - device: D7:B8:25:14:13:9B, auto: false registerApp()
 D/BluetoothGatt: registerApp() - UUID=d85876f0-1835-4163-a23e-12f2172ce284
 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=D7:B8:25:14:13:9B
 D/BluetoothGatt: discoverServices() - device: D7:B8:25:14:13:9B
 D/BluetoothGatt: onSearchComplete() = Device=D7:B8:25:14:13:9B Status=0
 D/BluetoothGatt: cancelOpen() - device: D7:B8:25:14:13:9B
 D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=D7:B8:25:14:13:9B
 D/Companion API: deviceDisconnected(): wearable device disconnected
 D/BluetoothGatt: close()`

I am doing stress tests to my system and try to connect/disconnect in loop for many iterations. This error is not reproducible, sometimes it raises and sometimes don't, do you have any clue of the possible error here? I cannot see it very clear in the logs. In general the library is working great, I am quite happy with it!

philips77, 2018-07-13:
onConnectionUpdated with interval 6 means that Android does service discovery. To make it as quick as possible, it switches the connection interval to lowest possible value. 6 means 7.5ms.
When you don't get it, it means that services were obtained from the cache and no service discovery we performed.

AvantStudioSL, 2018-07-13:
Thanks for the answer Aleksander! is there any option to configure using the cache or Android selects it internally?

philips77, 2018-07-13:
You'd have to clear the cache each time the device disconnects. There's a method shouldClearCacheAfter... in the BleManager v2.
But this will cost you some battery, as each time a phone connects, a full service discovery will be done.

AvantStudioSL, 2018-07-13:
Understood! I am going to try this way and check the battery consumption to see if it is acceptable for my project. Thanks again for the great support :)

closed 2018-08-13 by philips77