nordicsemi/Android-BLE-Library #18: Null pointer exception while turning off bluetooth.

midhun3112, 2018-07-10:
@philips77 
Version : **v2.0 alpha 5** 
I am using the Profile Service Ready Activity in the  nRF Toolbox/develop project. Everything is working as excepted but when I disconnect the bluetooth I am getting this error. Auto connect is turned off and also service is stopped after disconnect.

  java.lang.NullPointerException: Attempt to invoke virtual method 'void no.nordicsemi.android.ble.WriteRequest.notifyPacketSent(android.bluetooth.BluetoothDevice, byte[])' on a null object reference
        at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.onCharacteristicWriteSafe(BleManager.java:2348)
        at no.nordicsemi.android.ble.MainThreadBluetoothGattCallback.lambda$onCharacteristicWrite$3$MainThreadBluetoothGattCallback(MainThreadBluetoothGattCallback.java:119)
        at no.nordicsemi.android.ble.MainThreadBluetoothGattCallback$$Lambda$3.run(Unknown Source)
        at android.os.Handler.handleCallback(Handler.java:836)
        at android.os.Handler.dispatchMessage(Handler.java:103)
        at android.os.Looper.loop(Looper.java:203)
        at android.app.ActivityThread.main(ActivityThread.java:6269)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1063)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:924)

philips77, 2018-07-11:
Which profile are you testing?

midhun3112, 2018-07-11:
@philips77 I am using the BleProfileServiceReadyActivity in RSCActivity. After the disconnect, this crash occurs frequently if some data is pending. I am manually turning off the Bluetooth. Since there is no null check in the library for the onCharacteristicWriteSafe method the app crashes.

midhun3112, 2018-07-30:
@philips77 Any updates? :)

philips77, 2018-07-30:
Hi, I was on vacations and now I'm busy with another project right now and didn't have time to look into it. But I'll look into it when i have time.

midhun3112, 2018-07-30:
@philips77 Thanks for the quick reply :)

midhun3112, 2018-08-01:
@philips77 The crash is occurring on BleManager because when I turn off the Bluetooth the request queue is cleared But in onCharacteristicWriteSafe there is no null check for request
_final WriteRequest request = (WriteRequest) mRequest;
request.notifyPacketSent(gatt.getDevice(), data);_
This is where the crash occurs. Can you please check this and let me know.

		final void onCharacteristicWriteSafe(@NonNull final BluetoothGatt gatt,
											 @NonNull final BluetoothGattCharacteristic characteristic,
											 @Nullable final byte[] data,
											 final int status) {
			if (status == BluetoothGatt.GATT_SUCCESS) {
				log(Level.INFO, "Data written to " + characteristic.getUuid() +
						", value: " + ParserUtils.parse(data));

				onCharacteristicWrite(gatt, characteristic);
				final WriteRequest request = (WriteRequest) mRequest;
				request.notifyPacketSent(gatt.getDevice(), data);
				if (request.hasMore()) {
					enqueueFirst(mRequest);
				} else {
					mRequest.notifySuccess(gatt.getDevice());
				}
			} else if (status == BluetoothGatt.GATT_INSUFFICIENT_AUTHENTICATION
					|| status == 137 /* GATT AUTH FAIL */) {
				log(Level.WARNING, "Authentication required (" + status + ")");
				if (gatt.getDevice().getBondState() != BluetoothDevice.BOND_NONE) {
					// This should never happen but it used to: http://stackoverflow.com/a/20093695/2115352
					Log.w(TAG, ERROR_AUTH_ERROR_WHILE_BONDED);
					mCallbacks.onError(gatt.getDevice(), ERROR_AUTH_ERROR_WHILE_BONDED, status);
				}
                // The request will be repeated when the bond state changes to BONDED.
                return;
			} else {
				Log.e(TAG, "onCharacteristicWrite error " + status);
				mRequest.notifyFail(gatt.getDevice(), status);
				mValueChangedRequest = null;
				onError(gatt.getDevice(), ERROR_WRITE_CHARACTERISTIC, status);
			}
			nextRequest(true);
		}

philips77, 2018-08-01:
Yes, I'm going to add null checks is all callbacks.

closed 2018-08-13 by philips77