nordicsemi/Android-BLE-Library #19: API to set GAP event length

rahul-bedarkar, 2018-07-30:
Hi,

I was running Nordic  ATT MTU throughput test on nRF52840 between two boards. The test comes with SDK.

I want to do similar test with phone and nRF52840. I am using this library in my app. I can make connection to Nordic board which runs ATT MTU throughput.  I have Samsung S9 which has BT5. I can see that 2MBPS phy is set.  But it looks like there is no interface in this library to set GAP event length.

 NRF_SDH_BLE_GAP_EVENT_LENGTH - GAP event length
 The time set aside for this connection on every connection interval in 1.25 ms units.

Can you tell how to set this parameter from this android library ?

Also it looks like characteristic write is not synchronous. It means those are queued up. Is there any interface which does synchronous writes ?

Thanks,
Rahul

philips77, 2018-07-30:
Hi,
You may use requestMtu(...) to set the mtu (is it the gap event length?).
As to the write types, write with response should get callback after the ack was received, so it's synchronous. Write without response is in fact asynchronous and the callback is invoked after the requires was enqueued, but might have not even been started.

Are you using version 2 of this library? We will soon deprecate 1.x when 2 became stable.

rahul-bedarkar, 2018-07-30:
Hi,

Thanks for your quick reply.

I am using 1.x version. I am setting MTU with requestMTU(). But I need to set GAP event length. Could you suggest how to do that with this library?

I am using write without response, but I sometimes I don't get data on other side. It looks like phone didn't start sending yet.

Thanks,
Rahul

philips77, 2018-07-30:
There are just 3 things you may control with Android API. Mtu, connection priority (that is connection interval) and phy, on phones with Oreo+. 
To control the gap length you call requestConnectionPriority(..) where you provide a predefined value (read method's comment for details). This method on Android before 8 has no callback and you don't know how long time did it take, or has it succeeded. 
You may also request any other value from the peripheral.

rahul-bedarkar, 2018-07-30:
My Nordic board requests 7.5 msec connection interval which is 6 units. But app is only setting 9 units. This is happening regardless of requestConnectionPriority() call. Could you help why this might be happening? I want connection interval 6 units.

philips77, 2018-07-30:
Starting from Android 5 or 6 the lowest connection interval is 11.25 ms, set when you request priority high. Before it used to be 7.5 ms.
You may try to request 7.5 from the peripheral, but I guess phone will not allow.
7.5 ms is used only for service discovery, but then it goes up. This is to allow for more simultaneous connections, I guess.

closed 2018-08-13 by philips77