nordicsemi/Android-BLE-Library #23: Can't send 2 packages

Ne1c, 2018-08-16:
Hi, I send 2 packages but after the first package, flag mOperationInProgress became true and when I send the second package, package doesn't sending because mOperationInProgress == true. How I can fix it?

Version of library: 1.2.0

philips77, 2018-08-22:
Hi, 
Sorry for the late response. You may find an example how to do this here: https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/uart/UARTManager.java
The `mOperationInProgress` should automatically change to false after the `onCharacteristicWrite` callback was received. If it didn't, check if the characteristic you are writing to has correct properties (write or write no response) and whethere the data are actually being sent. You may check it and try writing with nRF Connect. When you are using Write With Response (aka Write, or Write Request), you may also use Long Write feature, but your device would have to support it.

Btw, I recommend switching to version2 (alpha 6 currently), as the 1.x version will soon become deprecated.

Ne1c, 2018-08-23:
I get response from a device after 2nd package and therefore callback doesn't call. I fixed this issue with delay in 200 ms.

philips77, 2018-08-23:
`onCharacteristicWrite` is called every time you write something to a characteristic. It's not related with notifications from the device.

Ne1c, 2018-08-31:
Ok, I resolved my problem. I just send every package after call `onCharacteristicWrite`

closed 2018-08-31 by Ne1c