nordicsemi/Android-BLE-Library #24: Build fail due to Invoke-customs are only supported starting with Android O (--min-api 26) Message{kind=ERROR, text=Invoke-customs are only supported starting with Android O (--min-api 26), sources=[Unknown source file], tool name=Optional.of(D8)}

rickymohk, 2018-08-27:
I got the following build error after using Android-BLE-Library
```Invoke-customs are only supported starting with Android O (--min-api 26)
Message{kind=ERROR, text=Invoke-customs are only supported starting with Android O (--min-api 26), sources=[Unknown source file], tool name=Optional.of(D8)}```
Then I change the Source Compatibility and Target Compatibility to 1.8 as suggested from the Internet
```    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }```
and got the following error
```Process 'command '/Applications/Android Studio.app/Contents/jre/jdk/Contents/Home/bin/java'' finished with non-zero exit value 1```
Is Android-BLE-Library requires a certain minSdkVersion? My project has minSdkVersion 19

philips77, 2018-08-27:
Our apps have minSdkVersion say to 19 as well. Do you have the correct jdk set up?

rickymohk, 2018-08-27:
I have been using the Android Studio embeded jdk, which is jersion 1.8.0_152. And I have also tried installing jdk1.8.0_181.jdk on my Mac from http://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html then use it as the JDK Location under Project Structure -> SDK Location. Both gives `finished with non-zero exit value 1` error. The `Invoke-customs are only supported starting with Android O (--min-api 26)` error appear as soon as I add `implementation 'no.nordicsemi.android:ble:1.2.0'` to dependencies and build the project, not even using any of it in the code.

rickymohk, 2018-08-28:
Solved by adding `android.enableD8.desugaring = true` to gradle.properties

closed 2018-08-28 by rickymohk