nordicsemi/Android-BLE-Library #25: Connection no longer possible after turning bluetooth off (when it is turned back on)

esensar, 2018-09-21:
This was tested on the develop branch (v2 library - 2.0-alpha6) and everything seems to be working correctly in all cases, except for when bluetooth is disabled on the device (after connection has been made). When bluetooth is re-enabled, connection is no longer possible.

I have debugged it and it seems that the issue is related to `mOperationInProgress` flag in `BleManagerGattCallback'. When call is made:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/135a7774e26d7bae61c13d584c8905c5c737fee5/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1778-L1786

flag is checked and call is silently skipped (none of the callbacks will be called in this case):
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/135a7774e26d7bae61c13d584c8905c5c737fee5/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L2739-L2747

When bluetooth is disabled on device, broadcast receiver is called:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/135a7774e26d7bae61c13d584c8905c5c737fee5/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L207-L249

which then prevents all further calls by setting this flag.
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/135a7774e26d7bae61c13d584c8905c5c737fee5/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L221-L224

I haven't seen it turned back to false to enable new connections.

philips77, 2018-09-21:
Could you check now? Or do you want me to release beta-2?

esensar, 2018-09-24:
Sorry for the late response, I haven't had the chance to test it before. It is still not working and I think that the issue is that bluetooth state receiver is unregistered once bluetooth is turned off, meaning the new code won't be called in case described in the issue.

`close()` is called once bluetooth is turned off:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/135a7774e26d7bae61c13d584c8905c5c737fee5/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L207-L249
and that method is unregistering bluetooth receiver:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/135a7774e26d7bae61c13d584c8905c5c737fee5/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L729-L736

philips77, 2018-09-24:
Oh, true. Let me think what can be done about this...

esensar, 2018-09-24:
Is there a specific reason why bluetooth state receiver is unregistered (since it is only doing something on bluetooth off state)?
Could that one be left registered? Others should be re-registered once device connects again.

philips77, 2018-09-24:
The initial intention was that you only call 'close()' or even 'disconnect()' and the manager is fully released. Connecting again registers the receiver again. But looks like i forgot about disabling BLE.

esensar, 2018-09-24:
But disabling BLE is currently different in that it sets the `mOperationInProgress` flag, which actually prevents further connections. Is the regular disconnect call disconnecting gracefully so it does not set these flags and does not clear the queue or? I am not sure if setting the flag is required, since queue is cleared too. I haven't analysed the library completely, so forgive me if I make some incorrect assumptions.

philips77, 2018-09-24:
You were right about unregistering the receiver. There's no need to handle STATE_ON them. Instead, I clear the `mOperationInProgress` flag. The reference to this object is nullified, but on next `enqueue()` it may be set again. Any thought? You need to listen for Bluetooth ON state on your own.

esensar, 2018-09-24:
Well, with the new fix I think it will clear up my issue. I need to test it now, I am not sure if there are some other use cases where something might go wrong. I don't think it should though, since whole queue is cleared and current calls are notified about the disconnection.

esensar, 2018-09-24:
I can confirm that this fix has resolved my issue. It would be nice if you could include this in beta2.
Thank you for the quick resolution

philips77, 2018-09-24:
I'll release in a second.

philips77, 2018-09-24:
Released in beta2.

philips77, 2018-11-06:
This needs verification again with 2.0-beta9. I had to revert one change that was introduced here.

philips77, 2018-11-06:
Fixed, tested and seems to be working.

closed 2018-11-06 by philips77