nordicsemi/Android-BLE-Library #26: autoConnect=true attempts to reconnect even after invoking disconnect()

maragues-kolibree, 2018-09-27:
I'm trying to disconnect from a device with a connection attempt in progress, but apparently the disconnect is never executed (I never receive any callback) and the _autoConnect=true_ causes an undesired reconnect attempt

Steps

1. Invoke _connect.await()_ on a switched off device
2. After some seconds, invoke 

```kotlin
        disconnect()
                .done { Timber.d("disconnect success") }
                .fail { device, status ->
                    Timber.d("disconnect fail")

                    close()
                }
                .enqueue()
```

3. After some time, _connect.await()_  throws a _RequestFailedException: Request failed with status 133_ and immediately attempts to reconnect. None of the disconnect callbacks is invoked

Is this working as intended or is it a bug?

maragues-kolibree, 2018-09-27:
This fixes the issue for now, but it's not pretty

This a wrapper that holds the _BleManager_ class

```kotlin
public void connect() {
    forceDisconnect.set(false);
    bleManager.setGattCallbacks(this);

    try {
      bleManager.connect(bluetoothDevice).await();
    } catch (RequestFailedException | DeviceDisconnectedException | BluetoothDisabledException e) {
      if (forceDisconnect.get()) {
        bleManager.setGattCallbacks(this);

        bleManager.close();
      }

      e.printStackTrace();
    }
  }

public void disconnect() {
    forceDisconnect.set(true);

    bleManager.disconnectWithoutReconnect();
  }
```

philips77, 2018-09-27:
Hi, I'll look into it. Thanks for the report!

philips77, 2018-09-28:
Hi @maragues-kolibree,
I was thinking about what should actually happen in this case. 
First of all, the correct way of closing pending connection attempt is by setting a timeout. This may be set wither using `enqueue(int)` or `await(int)`. The PR makes sure the connection gets cancelled in such case.
Also, the lib won't start another connection if the first one failed dues to a timeout.

maragues-kolibree, 2018-09-28:
If you release a new beta, I can test next Monday

philips77, 2018-10-01:
I released beta3.

philips77, 2018-10-01:
Fixed in beta4 :)

maragues-kolibree, 2018-10-01:
It's fixed, congrats and thanks!!

It's beta3, not beta4, just to keep track for the future.

closed 2018-10-01 by maragues-kolibree

maragues-kolibree, 2018-10-01:
After this change, is there a way to invoke _connect_ in a way that it keeps trying to reconnect after an error?

I know we can do it manually, just wondering.