nordicsemi/Android-BLE-Library #27: Parameter specified as non-null is null

maragues-kolibree, 2018-09-28:
The request that fails is

```kotlin
fun disconnectWithoutReconnect() {
        disconnect()
                .done { Timber.d("disconnect success") }
                .fail { device, status ->
                    Timber.d("disconnect fail")

                    close()
                }
                .enqueue()
    }
```

```
java.lang.IllegalArgumentException: Parameter specified as non-null is null: method kotlin.jvm.internal.Intrinsics.checkParameterIsNotNull, parameter it
        at com.mycompany.MyNordicBleManager$disconnectWithoutReconnect$1.onRequestCompleted(Unknown Source:2)
        at no.nordicsemi.android.ble.Request.notifySuccess(Request.java:829)
        at no.nordicsemi.android.ble.BleManager.internalDisconnect(BleManager.java:665)
        at no.nordicsemi.android.ble.BleManager.access$2100(BleManager.java:113)
        at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest(BleManager.java:2812)
        at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.access$1300(BleManager.java:1812)
        at no.nordicsemi.android.ble.BleManager.lambda$enqueue$4(BleManager.java:1786)
        at no.nordicsemi.android.ble.-$$Lambda$BleManager$KDSxdgn5nClZRFe_FvyLUfX8mWc.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6494)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
```

Scenario
- beta2
- attempt to disconnect from a device I'm not connected to. In this case, I never even attempted to connect to it.

philips77, 2018-09-28:
Thanks for testing the lib! I'll push sone updates asap.

philips77, 2018-09-28:
The `BluetoothDevice` could also be set in the BlueManager's constructor, instead of calling it in `connect(..)`. But that would break a lot when upgrading to the next version. What do you think?

maragues-kolibree, 2018-09-28:
From _disconnect_ javadoc

_Does nothing if device was not connected._

In this case, I'd expect the call to do nothing, since we are not connected. 

Does doing nothing also invoke _done_? I'm not sure if that's what I'd expect, but I understand that you'd rather invoke one of the two.

First option would be to remove the @NonNull annotation from BluetoothDevice and set it to @Nullable. That'd fix kotlin expectations, tho it's a breaking change that should be added to release notes.

As a second, more complex approach, would you consider an approach similar to RxJava's [Maybe](http://reactivex.io/RxJava/javadoc/io/reactivex/Maybe.html)? It either emits _onSuccess_, _onError_ or _onComplete_

From its javadoc: _onSuccess, onError and onComplete are mutually exclusive events_

_onComplete_'s name might be misleading and developers might think it's invoked after _done_, so maybe something like _ignored_ or _skipped_

Regarding the BluetoothDevice as a parameter to BleManager, that'd also break the possibility connect to multiple devices with a single BleManager. I don't know how desirable that is or if people are using it that way.

If it's not desirable to reuse a BleManager for multiple devices, then I'd definitely enforce it as a constructor parameter to avoid unexpected issues. Otherwise I'd leave it as it.

I'd maybe go for the 1st approach, adding a _ignored_ callback means increasing the complexity of the library. Is there any usecase for such a callback?

philips77, 2018-09-28:
I've already started implementing `invalid` callback to each request. It would only be called when `connect` was never called and `BluetoothDevice` is *null*. At least so far. That would also include Sleep request (which doesn't need `BluetoothDevice`, actually, but for consistency. And I left `@NonNull` in both `done` and `fail` callbacks. That way, I think, is the most "backwards compatible", as you don't need to make any changes, you don't need to handle the new `invalid` callback and you are guaranteed that `success` and `fail` will indeed have non-null parameter.

RxJava approach also seems good, but that would break everything.

Btw, how does the lib work with Kotlin? I tried to make it Kotlin-firendly, but actually never tried it with it.

maragues-kolibree, 2018-09-28:
Thanks for the quick response! 

There are no differences when using it from kotlin, It's straightforward.

The thing is, if a parameter is flagged with @Nonnull BluetootDevice, kotlin's type is _BluetoothDevice_, whereas if it's @Nullable the type is _BluetoothDevice?_, which is different from the non-question mark type. If a non-null parameter receives null, it's a runtime error because it's not of the type it expects.

philips77, 2018-09-28:
But there are those thing that make a library more Kotlin'y, like `@Nullable` or `@NonNull` that, that you already mentioned, that are converted to optional or not optional fields, but also callbacks as a last parameter can be open, etc. I just wonder if anything could be improved to make it better before going to final version.

maragues-kolibree, 2018-09-28:
sorry I misunderstood your question

I haven't found any issues, but I'm not that experienced with Kotlin

I know of two interop guides, but I guess you already checked that

https://kotlinlang.org/docs/reference/java-to-kotlin-interop.html
https://android.github.io/kotlin-guides/interop.html

philips77, 2018-09-28:
Ok, this issue should be fixed with the new PR. Would you like to have a look at it before I merge?

philips77, 2018-10-01:
Released in beta3.

maragues-kolibree, 2018-10-01:
It's fixed, congrats and thanks!!

closed 2018-10-01 by maragues-kolibree