nordicsemi/Android-BLE-Library #29: [Beta3] NPE BleManager$BleManagerGattCallback.nextRequest(boolean)' on a null object reference

maragues-kolibree, 2018-10-01:
On Beta3, it happened after I tried to connect and received a _TimeoutException_

```
Process: com.mycompany, PID: 20828
    java.lang.NullPointerException: Attempt to invoke direct method 'void no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest(boolean)' on a null object reference
        at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.access$1300(BleManager.java:1836)
        at no.nordicsemi.android.ble.BleManager.lambda$enqueue$4(BleManager.java:1804)
        at no.nordicsemi.android.ble.-$$Lambda$BleManager$KDSxdgn5nClZRFe_FvyLUfX8mWc.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6494)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)
```

philips77, 2018-10-01:
There is no callback for `close()`. What exactly did you do? You actually don't need to call `close()`, it does it automatically after you disconnect/fail to connect.

maragues-kolibree, 2018-10-01:
I was referring to those BluetoothGatt log lines. I edited the description with more details.

I'm still occasionally seeing the NPE when attempting to retry a connection to a disconnected device, even after introducing a 600ms delay

philips77, 2018-10-01:
Does it happen on error 133? Do you try to reconnect to the device there, or what are you doing? Could you, please, post some of your code related to this issue?

philips77, 2018-10-01:
You are still trying to connect to a switched-off device, yes? And time-outing after 30 sec, yes?

maragues-kolibree, 2018-10-01:
Yep, that's the scenario

philips77, 2018-10-02:
Hi, could you load the BLE library from the project instead of jcenter? I could send you the BleManager.java file for testing. Ok?

philips77, 2018-10-02:
Please, check out branch *bugfix/29*.

maragues-kolibree, 2018-10-02:
I'll try to do it later today 👍

philips77, 2018-10-02:
Hi, I've added some more features to the mentioned branch, including automatic connection retries with delay. I've just pushed my recent changes. Did you have time to check the fix?

maragues-kolibree, 2018-10-02:
I'm going to do it right now