nordicsemi/Android-BLE-Library #30: createBond returning twice

Rawa, 2018-10-01:
Hello,

I'm using we are using your library for managing my BT connections and mostly it is working great. But I seem to run into some problems with the bonding process. It seems like the enqueue gets two callbacks, done and fail. 

In short what happens is, I connect, create a bond, get success (done). I then disconnect and remove the bond and receive a fail.

I added a random number to log to just clarify I'm not calling the function twice.

```
2018-10-01 16:19:05.047 623-1923/com.pansaraugmented.app D/RxBt: waitForBond 0.6858416588166597
2018-10-01 16:19:05.053 623-623/com.pansaraugmented.app I/BluetoothDevice: createBond() for device 91:3 called by pid: 623 tid: 623
2018-10-01 16:19:05.126 623-623/com.pansaraugmented.app E/RxBt: Bonding required
2018-10-01 16:19:06.250 623-623/com.pansaraugmented.app E/RxBt: Bonded!
2018-10-01 16:19:06.251 623-623/com.pansaraugmented.app D/RxBt: Already Bond success 0.6858416588166597
2018-10-01 16:19:24.629 623-623/com.pansaraugmented.app D/BluetoothDevice: bond, retrieved method: public boolean android.bluetooth.BluetoothDevice.removeBond()
2018-10-01 16:19:24.630 623-623/com.pansaraugmented.app I/BluetoothDevice: removeBond() for device 91:3 called by pid: 623 tid: 623
2018-10-01 16:19:24.687 623-623/com.pansaraugmented.app D/RxBt: Already Bond Fail 0.6858416588166597
```

Rawa, 2018-10-01:
I can try and produce a minimal reproducible example if needed.

philips77, 2018-10-02:
Hi, could you try the version from branch *bugfix/29*? You need to add the library as module, not from jcenter"

In **build.gradle**:
```groovy
    // api 'no.nordicsemi.android:ble:2.0-beta4'
    api project(":ble")
```
In **settings.gradle**:
```groovy
include ':ble'
project(':ble').projectDir = file('../Android-BLE-Library/ble')
```
*../Android-BLE-Library/ble* is the path to *ble* module.

philips77, 2018-10-02:
Btw, thanks for testing the library!

closed 2018-10-02 by philips77