nordicsemi/Android-BLE-Library #33: ConnectRequest with autoConnect=true does not attempt to reconnect after timeout

maragues-kolibree, 2018-10-04:
Version: Beta5
**Scenario**
- attempt to connect to an offline device. After the connection attempt times out, switch the device on

```java
bleManager.connect(bluetoothDevice)
      .useAutoConnect(true)
      .retry(3, 600)
      .enqueue();
```

**Expected**
The connection is attempted after the timeout

**Got**
The connection is never attempted after the timeout

**Logs**
```
13:58:41.993 com.mycompany D/BluetoothGatt: connect() - device: C0:4B:24:4B:11:A7, auto: false
13:58:41.993 com.mycompany D/BluetoothGatt: registerApp()
13:58:41.993 com.mycompany D/BluetoothGatt: registerApp() - UUID=1cc136ae-1d6f-4633-894a-5cf88ce20c60
13:58:41.995 com.mycompany D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7

13:58:41.992 com.mycompany E/MyBleManagerCallback: onDeviceConnecting: C0:4B:24:4B:11:A7 started connecting to the TB

13:59:12.002 com.mycompany D/BluetoothGatt: onClientConnectionState() - status=133 clientIf=7 device=C0:4B:24:4B:11:A7
13:59:12.016 com.mycompany D/BluetoothGatt: close()
13:59:12.016 com.mycompany D/BluetoothGatt: unregisterApp() - mClientIf=7

13:59:12.021 com.mycompany E/MyBleManagerCallback: onDeviceDisconnected: C0:4B:24:4B:11:A7 disconnected
13:59:12.027 com.mycompany E/MyBleManagerCallback: onError: a BLE error has occurred message = Error on connection state change, code = 133
```

Related https://github.com/NordicSemiconductor/Android-BLE-Library/pull/31#issuecomment-426665336

philips77, 2018-10-04:
But that's intended. As `retry(...)` method's docs says: https://github.com/NordicSemiconductor/Android-BLE-Library/blob/8d3ddf478c93846d6b06586116a59f9042768f53/ble/src/main/java/no/nordicsemi/android/ble/ConnectRequest.java#L118-L132
The correct way of solving this would be to scan before connecting to make sure the device is reachable. Check out #15 at the end.

philips77, 2018-10-04:
(I actually clarified the docs yesterday in here 8d3ddf478c93846d6b06586116a59f9042768f53)