nordicsemi/Android-BLE-Library #35: BleManagerGattCallback.nextRequest stucks after onBonded

sztomek, 2018-10-12:
Hi,

I ran into the mentioned issue using version 1.2.0 of the library. The app is running on a Nexus 5 with Android 8.1.
I'm trying to connect to a nRF52840 dongle, then bond. After bonding I want to change MTU size and write some characteristics on the device.

The flow in details:
- after device discovery, I call `BleManager.connect(BluetoothDevice)`
- in my `BleManagerGattCallback.initGatt` implementation, I push a single bond request (`Request.createBond()`) into the requests queue
- `BleManageCallbacks.onBondingRequired(BluetoothDevice)` is invoked, Android pops up the Pairing notification, where I choose to pair. In the nRF Connect app, I also confirm that the pins match.
- `BleManagerCallbacks.onBonded(BluetoothDevice)` is invoked. In my implementation I enqueue a new MTU change request here: `enqueue(Request.newMtuRequest(128))`

The request never gets processed since the `BleManager.BleManagerGattCallback.mOperationInProgress` flag is still `true`.

I actually checked out the source of the library and modified the `BleManager` implementation to reset the `mOperationInProgress` flag to `false` on successful bonding in the `BroadcastReceiver`'s `BOND_BONDED` case:
					
``` java
                                        Logger.i(mLogSession, "Device bonded");
					mGattCallback.mOperationInProgress = false; // added this line
					mCallbacks.onBonded(device);
					// If the device started to pair just after the connection was established the services were not discovered.
					if (mBluetoothGatt.getServices().isEmpty()) {
						mHandler.post(() -> {
							Logger.v(mLogSession, "Discovering Services...");
							Logger.d(mLogSession, "gatt.discoverServices()");
							mBluetoothGatt.discoverServices();
						});
						return;
					}
					break;

```
With the modifications above I managed to run the MTU change and other requests after bonding.

Is my solution valid or did I do something wrong in my intitial flow?

Thanks in advance!

philips77, 2018-10-12:
Hi, your fix looks good. Actually, we are closing migration to version 2 of the library, where a lot has been improved. I can recall fixing this issue there (but you may not recognize some of the code). Yesterday I also fixed(?) an issue with bonding there, and will release a version 2.0-beta6 when #32 is confirmed to work.
I would recommend you to switch to the new version if the deadline is far ahead. If not, your fix is OK. Would you like me to release version 1.2.1, or can you work on your fork?

philips77, 2018-10-12:
I would say it's much more stable and tested then 1.2, at least by myself. Thanks to those who already migrated I also managed to fix some bugs that I didn't see before, so any feedback is welcome. There will for sure be beta6 with all the recent commits (hopefully today, depending on the #32 status). I'm not aware of any more bugs in it and want to release it as soon as ASAP. Let's say the release date to be **next week**.

nRF Toolbox and nRF Blinky and the BLE Common Library also are waiting to go out of *develop* branch.

philips77, 2018-10-22:
Hi, sorry for the delay, I was sick. I'm ready to release the 2.0. Did you have time to look at it? Do you have any comments?

sztomek, 2018-10-30:
Hi @philips77 

Sorry for the very late answer. 
Yup, I have updated the verison of the lib to 2.0.0-beta5 and tried to bond after connection. Unfortunately I experienced the same behavior as before: the request queue stucks. 

I will check again with the stable v2.0.0 tomorrow and will provide more details if the issue still exists.

sztomek, 2018-10-31:
Hi again,

Just checked the latest version (2.0.0-beta7) and the scenario I have described earlier works just fine.
So I assume this ticket can be closed.

Thanks!

philips77, 2018-10-31:
I'll leave it here until the version 2 goes final. In theory I should fix it in 1.2, but I have no time to do so now :(

stefnotch, 2021-03-05:
Looks like version 2 is out now. Should this ticket be closed now?

closed 2021-03-05 by philips77