nordicsemi/Android-BLE-Library #36: [beta6] NPE invoking _mGattCallback.nextRequest_

maragues-kolibree, 2018-10-23:
I received a crash stacktrace that points to BleManager.enqueue

```
	protected final void enqueue(@NonNull final Request request) {
		if (mGattCallback == null) {
			mGattCallback = getGattCallback();
		}
		// Add the new task to the end of the queue.
		mGattCallback.enqueue(request);
		runOnUiThread(() -> mGattCallback.nextRequest(false)); <-- NPE
	}
```

When executing that Runnable, mGattCallback is null. It obviously wasn't null in the line above, which means we nullified the gattCallback instance. If I'm not mistaken, this can only happen due to a _close_ invocation.

The easiest way is to add a null check on mGattCallback after locking _mLock_. If you are fine with this solution, I can create the PR, but you may want to analyze if there are other consequences.

philips77, 2018-10-23:
It's already fixed in version 2:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/develop/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1853 and was released in 2.0-beta6. Commit: 17096338124604c9ac1edb3cee57e2d9d34aa067

philips77, 2018-10-23:
Are you talking about version 1.2 of the library?

maragues-kolibree, 2018-10-23:
I thought I was on beta6 but I was still on beta5, sorry

closed 2018-10-23 by maragues-kolibree