nordicsemi/Android-BLE-Library #37: [Beta6] NoSuchElementException

maragues-kolibree, 2018-10-25:
Device: Nexus 4 running 5.1.1

```
    java.util.NoSuchElementException
        at java.util.LinkedList.removeFirstImpl(LinkedList.java:689)
        at java.util.LinkedList.removeFirst(LinkedList.java:676)
        at java.util.LinkedList.poll(LinkedList.java:895)
        at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest(BleManager.java:2864)
        at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.onCharacteristicWriteSafe(BleManager.java:2482)
        at no.nordicsemi.android.ble.MainThreadBluetoothGattCallback.lambda$onCharacteristicWrite$3(MainThreadBluetoothGattCallback.java:128)
        at no.nordicsemi.android.ble.-$$Lambda$MainThreadBluetoothGattCallback$dAnXisth7CAlJFFhbDJ5X_192I0.run(lambda)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:135)
        at android.app.ActivityThread.main(ActivityThread.java:5254)
        at java.lang.reflect.Method.invoke(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:372)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:903)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:698)```

maragues-kolibree, 2018-10-25:
It fails consistently on Nexus4, but the same operations on other devices never trigger this issue

philips77, 2018-10-30:
This indeed is very interesting. The `poll()` should return `null` if the list is empty:
https://android.googlesource.com/platform/libcore/+/lollipop-mr1-release/luni/src/main/java/java/util/LinkedList.java#895

But if `size != null`, then it calls `removeFirst()` which actually may throw an exception in that case:
https://android.googlesource.com/platform/libcore/+/lollipop-mr1-release/luni/src/main/java/java/util/LinkedList.java#713
The question is, why the list is empty and has size > 0, or where does it get messed up. Looks like race condition.

I tried running nRF Blinky on Nexus 4 with Android 5.1.1, but didn't have problem like this. Do you add or remove tasks in some weird way in your app?
The only solution I can find is to enclose it all in try-catch to avoid crashes like this.

maragues-kolibree, 2018-10-31:
I'm enqueuing multiple write requests at the same time. I don't see any failure before this NPE.

I'm thinking this is a bug on my side, they should be queued one after the other, but it's working on most devices.

philips77, 2018-10-31:
I've released 2.0-beta7 with all the recent changes (except Reliable Write).

closed 2018-11-05 by philips77