nordicsemi/Android-BLE-Library #38: ClassCastException: Bootstrap method returned null

eliotstock, 2018-10-26:
When using beta6, trying to connect to a device from the onScanResult() method in the scan callback as follows:

        BleDeviceManager bleDeviceManager = BleDeviceManager.getInstance(context.getApplicationContext());

        BleDevice2 device = new BleDevice2();
        bleDeviceManager.setGattCallbacks(device);

        try {
            bleDeviceManager.connect(result.getDevice())
                    .retry(3, 100)
                    .useAutoConnect(false)
                    .enqueue();
        }
        catch (Exception e) {
            Logg.e(TAG, "Can't connect to device: " + e.getMessage());
        }

I get the following stack trace:

```
2018-10-26 17:06:47.480 20222-20222/cc.biketracker.android I/BleScanCallback: Scan result: Device name: 00228892, address: C7:56:60:F0:56:58, RSSI: -59
2018-10-26 17:06:47.493 20222-20222/cc.biketracker.android D/BleScanCallback: Bond state: BOND_BONDED
2018-10-26 17:06:47.498 20222-20222/cc.biketracker.android D/BleScanCallback: Found device. Stopping scanning.
2018-10-26 17:06:47.503 20222-20222/cc.biketracker.android D/BluetoothAdapter: isLeEnabled(): ON
2018-10-26 17:06:47.543 20222-20222/cc.biketracker.android D/AndroidRuntime: Shutting down VM
2018-10-26 17:06:47.558 20222-20222/cc.biketracker.android E/AndroidRuntime: FATAL EXCEPTION: main
    Process: cc.biketracker.android, PID: 20222
    java.lang.BootstrapMethodError: Exception from call site #9 bootstrap method
        at no.nordicsemi.android.ble.BleManager.enqueue(BleManager.java:1860)
        at no.nordicsemi.android.ble.Request.enqueue(Request.java:751)
        at cc.biketracker.android.ble.BleScanCallback.onScanResult(BleScanCallback.java:173)
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:484)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6680)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
     Caused by: java.lang.ClassCastException: Bootstrap method returned null
        at no.nordicsemi.android.ble.BleManager.enqueue(BleManager.java:1860) 
        at no.nordicsemi.android.ble.Request.enqueue(Request.java:751) 
        at cc.biketracker.android.ble.BleScanCallback.onScanResult(BleScanCallback.java:173) 
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:484) 
        at android.os.Handler.handleCallback(Handler.java:873) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:193) 
        at android.app.ActivityThread.main(ActivityThread.java:6680) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858) 
2018-10-26 17:06:47.565 20222-20222/cc.biketracker.android E/UncaughtException: java.lang.BootstrapMethodError: Exception from call site #9 bootstrap method
        at no.nordicsemi.android.ble.BleManager.enqueue(BleManager.java:1860)
        at no.nordicsemi.android.ble.Request.enqueue(Request.java:751)
        at cc.biketracker.android.ble.BleScanCallback.onScanResult(BleScanCallback.java:173)
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:484)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:6680)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858)
     Caused by: java.lang.ClassCastException: Bootstrap method returned null
        at no.nordicsemi.android.ble.BleManager.enqueue(BleManager.java:1860) 
        at no.nordicsemi.android.ble.Request.enqueue(Request.java:751) 
        at cc.biketracker.android.ble.BleScanCallback.onScanResult(BleScanCallback.java:173) 
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:484) 
        at android.os.Handler.handleCallback(Handler.java:873) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:193) 
        at android.app.ActivityThread.main(ActivityThread.java:6680) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:858) 
```

It could be I'm using the API incorrectly. It seems strange that this is supposed to happen on the UI thread, given I'm scanning in the background from a service. I did try using await() rather than enqueue, but that failed with another exception indicating I *should* be using enqueue().

philips77, 2018-10-27:
What does your getGattCallbacks() method return?

philips77, 2018-10-27:
What is this reception that you're getting? Does it happen because of npe I'm the library?
Await should be called from non-ui thread, enqueue from any thread. All BLE operations are executed in main (ui) thread. All callbacks are also invited in ui thread.

eliotstock, 2018-10-27:
I think it may actually be because this is the first library I'm using in my app with JDK 1.8 features (you have a lambda in your code). I had this in my gradle file:

    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_7
        targetCompatibility JavaVersion.VERSION_1_7
    }

I'll test tomorrow with this:

    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }

eliotstock, 2018-10-31:
Now I can't repro this, and nothing's really changed. It could be that I just needed another device reboot or an Android Studio restart. Sorry for the false alarm.

closed 2018-10-31 by eliotstock

JoshuaRack, 2019-09-29:
I also had this issue, fixed by enabling java 8 for my app module, even though it was enabled for this module. A note in the readme might be useful for others.