nordicsemi/Android-BLE-Library #39: getGattCallback null object

cirlam, 2018-10-30:
Hi,

I've had 2 of my users recently crash with the following null object crash on beta5 version of the library. I'm unsure what exactly is happening at the moment in time before it crashes as I've never seen this on my devices, but wanted to share it in case there is something obvious you can spot? 
I have several hundred other users all using the app (and library) without problem, so don't think this is a huge problem. 

Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'void no.nordicsemi.android.ble.a$a.a(android.os.Handler)' on a null object reference
       at no.nordicsemi.android.ble.BleManager.getGattCallback(BleManager.java:680)
       at no.nordicsemi.android.ble.BleManager.getGattCallback(BleManager.java:114)
       at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.lambda$onConnectionStateChangeSafe$1(BleManager.java:2237)
       at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.lambda$hHV_dPbGJHyD-KM5FND5GY7KZA0(Unknown Source)
       at no.nordicsemi.android.ble.-$$Lambda$a$a$hHV_dPbGJHyD-KM5FND5GY7KZA0.run(Unknown Source:4)
       at android.os.Handler.handleCallback(Handler.java:789)
       at android.os.Handler.dispatchMessage(Handler.java:98)
       at android.os.Looper.loop(Looper.java:164)
       at android.app.ActivityThread.main(ActivityThread.java:6944)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)

philips77, 2018-10-30:
From what I can see, this happens when you call `manager.close()` just after enqueing a connect request. When the request gets proceed in UI thread, the `mGattCallbacks` will be `null`.

cirlam, 2018-11-07:
Thank you  - this seems ok (no crashes seen for the past week) with your latest fix :)