nordicsemi/Android-BLE-Library #42: Getting Android to generate LL_LENGTH_REQ, to extend PDU

readysteadygo2006, 2018-11-06:
Hi,

I'm trying to use the library to get communication between an Android mobile and a BLE peripheral
which utilises the extended PDU.

To understand things I wrote the central using the Nordic SDK 15.x. With the Nordic central I found it was
necessary to get the central to send a LL_LENGTH_REQ, to get the peripheral to extend the length of
the PDU. This was required in addition to the request to increase the MTU size.

Does the library support making such a request from an Android device?

Is it possible to make this request from and Android device?

I'm not certain whether this is the real problem, as there seems to be discussion about
Android transparently chunking characteristics over 20 bytes length. Is it also necessary
to tell Android not to employ this mechanism, to ensure it utilises the full PDU
once its enabled?

I'm noticing that currently extended length notification is being transferred in fragments.

Again is the control of the chunking within Android something the library supports?

Grateful for any help you can provide on this matter!

philips77, 2018-11-06:
Hi, the Data Length Extension (DLE), a feature from Bluetooth 4.2 (as I understand this is the LL_LENGTH_PDU, am I correct?) is supported on newer phones with Android 6*, as far as I remember. There is no any API to enable or disable it. Android will use it if MTU > 23 and both sides support it. Android only allows to request MTU, which the library is doing, as this is now supported by the DFU bootloader since SDK 15.

\* Newer phones = some phones got update to Android 6 and they may not support it.

readysteadygo2006, 2018-11-07:
Hi ,

This is good news!

However my peripheral requires the central to explicitly request that the PDU length be extended via
a LL_LENGTH_REQ. 

My peripheral is mbed based, however would a Nordic peripheral no also require such a request before it would extend the PDU length?

philips77, 2018-11-07:
For this question I have to redirect you to DevZone. Or support team know this stuff.

ashokkandasamy, 2020-11-23:
@philips77  has his been answered in DevZone, can you share any reference. 

We were using NRF52 - would like to know - from iOS/Android phones how to do MTU negotiation v/s PDU size.