nordicsemi/Android-BLE-Library #43: Callback onConnectionStateChange after disconnect can put BleManager in inconsistent state

al-broco, 2018-11-06:
I am seeing a situation where `onConnectionStateChangeSafe` in `BleManager` gets called with `newState == BluetoothProfile.STATE_CONNECTED` after I do a disconnect (call `BleManager.disconnect()`).

This causes `mConnected` to be set to true, but `mBluetoothDevice` is null. This makes both connect do nothing (since `mConnected` is true) and disconnect to nothing (since `mBluetoothDevice` is null), so the BleManager is no longer of much use.

Some more rough details on what I'm doing:
1. Connecting (calling `BleManager.connect()`).
2. Enabling notifications for a couple of characteristics. Reading and writing some characteristics.
3. Disconnecting.
4. Repeat.

This is not done in quick succession - between the disconnect and the next connect for instance there is a pause several seconds long.

This works fine most of the time, but sometimes, there us a call to `onConnectionStateChangeSafe` after the disconnect, which causes the problem described above.

Side note: The purpose for this is to implement a communications protocol where characteristics are used to transfer data, and the error handling is specified as disconnecting, reconnecting, and starting over.

I have been using version 2.0-beta10.

philips77, 2018-11-07:
Thank you for the feedback. I can't do anything about the extra `onConnectionStateChangeSafe`, as it's being called by the system, but the lib should not get stack when it's called. Expect a fix soon.

philips77, 2018-11-07:
From my experience this happens when Android receives a notification after the `disconnect()` was called. First it notifies you that it's disconnected, but when it gets the notification it's like "hmm.. I'm still connected, so here's your callback with connected state".
What phone are you using for testing and how often a) you send notifications from the peripheral b) you experience this issue? You are not using `autoConnect`, right?

There are 2 solutions that can be done here:
1. In `onConnectionStateChangedSafe`, where `status == BluetoothGatt.GATT_SUCCESS && newState == BluetoothProfile.STATE_CONNECTED`, and `mBluetoothDevice == null` I may call `gatt.close()` before setting `mConnected` to *true* and pretend that it never happened. After all, you called `disconnect()`, so the desire effect is to be disconnected.
2. I may try to recover to a state from before disconnection: `mBluetoothDevice = gatt.getDevice()`, `mBluetoothGatt = gatt` , and `mGattCallbacks = this`. But I'd have to register the receivers again, as `close()` was called before.

Looks like 1. is a better solution.

al-broco, 2018-11-07:
Thanks for the quick fix! Here's the info you asked for:
Phone used for testing: Google Pixel with Android 7.1.2.
a. When doing this test, constantly. The peripheral is trying to stream data by repeatedly writing a characteristic.
b. Fairly frequently, maybe after 20% of the disconnects (rough approximation). No, I am not using autoConnect.

al-broco, 2018-11-08:
I have now, the fix seems to have solved the problem. I have run the same test with the new code 50 times, no issues, connection works beautifully. Thanks again!

closed 2018-11-08 by philips77