nordicsemi/Android-BLE-Library #52: Task: Migrate to AndroidX

philips77, 2018-11-30:
The Android Jetpack is stable now. This lib would only use the annotations from AndroidX.
Importing the lib as a module to projects already migrated requires the lib also to be migrated.

eliotstock, 2018-12-15:
I just realised how badly I need this ;-) Is v2.1 far off, do you think? Is early Jan possible?

As a workaround, I can use the old pre-Android X version of ViewModel and LiveData in the meantime, but would prefer to be on the new one, of course.

    // implementation 'androidx.lifecycle:lifecycle-extensions:2.0.0'
    implementation 'android.arch.lifecycle:extensions:1.1.1'

philips77, 2018-12-16:
It's a simple task. But I wanted to add some more features in 2.1. But they can come on 2.2 :)

PR #69 (2018-12-18): Migration to Android X
The time has come. This PR resolves #52.
closed 2018-12-18 by philips77