nordicsemi/Android-BLE-Library #58: NullPointerException processing disconnect

chris-prop, 2018-12-10:
We are sometimes getting the following stack trace if we turn off bluetooth while communicating with a device:

`no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest(boolean)' on a null object reference
        at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.access$1300(BleManager.java:2071)
        at no.nordicsemi.android.ble.BleManager.internalDisconnect(BleManager.java:869)
        at no.nordicsemi.android.ble.BleManager.onRequestTimeout(BleManager.java:2056)
        at no.nordicsemi.android.ble.TimeoutableRequest.lambda$notifyStarted$0(TimeoutableRequest.java:188)
        at no.nordicsemi.android.ble.-$$Lambda$TimeoutableRequest$LRW3H3z9m1CuZHgzj67Qxd_-s8Y.run(lambda)
        at android.os.Handler.handleCallback(Handler.java:751)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6816)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1563)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1451)`

chris-prop, 2018-12-12:
@philips77 It looks like this is happening when disabling bluetooth on the phone while communicating with a device.

Do you have a timeframe for an update?

Would a PR help? I see some null checks in other parts of the code. Any thought if that would be a complete fix?

Thanks

philips77, 2018-12-12:
I'll look at it on Friday, ok?

philips77, 2018-12-17:
Released in 2.0.4.

chris-prop, 2018-12-20:
@philips77 Thanks for your work on this. I see you commented it is released in 2.0.4. This version shows on the main page, but is not listed under releases. Is the release final?

philips77, 2018-12-21:
Sorry, I'll release it immediately.