nordicsemi/Android-BLE-Library #59: timeout params has some problem

qq1012298028, 2018-12-11:
When I use the timeout in connect, the connection becomes extremely difficult, and some devices can't even connect.
version is v2.0.3

qq1012298028, 2018-12-11:
oh! Is TimeOut Method

philips77, 2018-12-11:
Hello,
What is the issue exactly? Mind, that the `timeout` parameter you set in `.timeout(timeout)` is in milliseconds. Android will by default timeout a connection attempt (with autoConnect=false) in 30 seconds, reporting error 133. So for `ConnectRequest` it only makes sense to set the timeout to be less then 30 sec (30000 ms).

qq1012298028, 2018-12-11:
I set .timeout(15000)  and autoConnect=true,     but connect is very difficult.then I remove .timeout(timeout),It has better performance

qq1012298028, 2018-12-13:
I set userAutoConnect(false),is can work. Thanks

qq1012298028, 2018-12-14:
Ok, I understand, thank you very much for your Lib and help.