nordicsemi/Android-BLE-Library #61: NPE Connect notifySuccess

maragues-kolibree, 2018-12-13:
I had two phones competing to connect to the same device, and this crash happened in one of them. The other did connect successfully

```
java.lang.NullPointerException: Attempt to invoke virtual method 'void no.nordicsemi.android.ble.ConnectRequest.notifySuccess(android.bluetooth.BluetoothDevice)' on a null object reference
        at no.nordicsemi.android.ble.BleManager.internalConnect(BleManager.java:733)
        at no.nordicsemi.android.ble.BleManager.access$2100(BleManager.java:113)
        at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.onConnectionStateChangeSafe(BleManager.java:2495)
        at no.nordicsemi.android.ble.MainThreadBluetoothGattCallback.lambda$onConnectionStateChange$0(MainThreadBluetoothGattCallback.java:107)
        at no.nordicsemi.android.ble.-$$Lambda$MainThreadBluetoothGattCallback$nMa21BAwq4uT_MjxPntqE-NM9y8.run(lambda)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:145)
        at android.app.ActivityThread.main(ActivityThread.java:6939)
        at java.lang.reflect.Method.invoke(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:372)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1404)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1199)
```

version 2.0.3

philips77, 2018-12-17:
Released in 2.0.4.

closed 2018-12-17 by philips77

zippert, 2022-06-09:
Bug still exists on version 2.4.1:

```
Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean no.nordicsemi.android.ble.ConnectRequest.notifySuccess(android.bluetooth.BluetoothDevice)' on a null object reference
       at no.nordicsemi.android.ble.BleManagerHandler.internalConnect(BleManagerHandler.java:489)
       at no.nordicsemi.android.ble.BleManagerHandler.access$2700(BleManagerHandler.java:57)
       at no.nordicsemi.android.ble.BleManagerHandler$3.lambda$onConnectionStateChange$4$no-nordicsemi-android-ble-BleManagerHandler$3(BleManagerHandler.java:1810)
       at no.nordicsemi.android.ble.BleManagerHandler$3$$ExternalSyntheticLambda8.run(:4)
       at android.os.Handler.handleCallback(Handler.java:938)
       at android.os.Handler.dispatchMessage(Handler.java:99)
       at android.os.Looper.loop(Looper.java:223)
       at android.app.ActivityThread.main(ActivityThread.java:7888)
       at java.lang.reflect.Method.invoke(Method.java)
       at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:981)
```

Please advice