nordicsemi/Android-BLE-Library #63: Add filter to WaitForValueChangedRequest

maragues-kolibree, 2018-12-14:
With `WaitForValueChangedRequest`, we can write on a characteristic and wait for the response

```kotlin
waitForNotification(responseCharacteristic)
                    .trigger(writeCharacteristic(gattCharacteristicWrite, payload))
                    .awaitValid(response)
```

The issue is that it simply returns the first value notified on `responseCharacteristic`, and there's no way to set a filter

In our case, there can be multiple writes on the same characteristic, but the response will always contain the commandId at byte[0]. So, this'd be my feature request

```kotlin
waitForNotification(responseCharacteristic)
                    .trigger(writeCharacteristic(gattCharacteristicWrite, payload))
                    .filter(response -> response[0] == expectedCommandId)
                    .awaitValid(response)
```

`filter` could take a `Predicate<Boolean>`, for example

Furthermore, there could be multiple operations enqueued writing the same commandId, and they should deliver the results in a FIFO way

Could this be a problematic scenario?
1. Read name -> "1 name"
2. Set name -> "2 name"
3. Read name -> ??

Depending on the state of the queue, could the 3rd operation could receive either "1 name" or "2 name"?

I'm not sure if it'd be problematic or not, considering there's a single queue, maybe it'd always be ok.

philips77, 2018-12-14:
Did you try using [`merge(DataMerger)`](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/WaitForValueChangedRequest.java#L124) for purpose of filtering?

https://github.com/NordicSemiconductor/Android-BLE-Library/blob/09e7adb6a93354cf07392285572a3d072dadf989/ble/src/main/java/no/nordicsemi/android/ble/data/DataMerger.java#L31-L41

From this method you could return true only if the right message was received:

```java
boolean merge(@NonNull final DataStream output,
			  @Nullable final byte[] lastPacket, @IntRange(from = 0) final int index) {
   if (lastPacket != null && lastPacket.length > 0 && lastPacket[0] == SOME_COMMAND_ID) {
      stream.write(lastPacket);
      return true;
   }
   return false;
}

It wasn't designed for this purpose, but I guess it could be used here. Of course your filtering method would be nicer.
```

maragues-kolibree, 2018-12-14:
That seems to work 👏 

Would it make sense to create a PR adding a `filter` that does this under the hood?

philips77, 2018-12-14:
You should be able to set filter AND the merger (the merger would merge only those packets that match filter). So implementing filter in the library must use some additional code, not `merge`. But the code could look very alike. 
What would your vision of the API using `Predicate<Boolean>`?

maragues-kolibree, 2018-12-14:
This is how I would like to use the API, if that was your question (c&p from first message)

```waitForNotification(responseCharacteristic)
                    .trigger(writeCharacteristic(gattCharacteristicWrite, payload))
                    .filter(response -> response[0] == expectedCommandId)
                    .awaitValid(response)```

I'm not sure if this was your question.

I'm only thinking about `WaitForValueChangedRequest`, I don't know if it makes sense in other requests

Maragues, 2018-12-19:
I'll test the fix when version 2.1 is released.

We recently migrated to `includeBuild` and I'm having issues referencing a project in my filesystem, sorry

philips77, 2019-02-01:
Released in 2.1.0.

closed 2019-02-01 by philips77