nordicsemi/Android-BLE-Library #70: Connect hangs on some devices

nicrgren, 2018-12-17:
Hi,

Using your scanner compat library I've noticed that the scanner sometimes discovers devices where the record.device.name is null. Whenever the scanner identifies one of your boards like this I cannot connect to it - BleManager.connect(device) simply times out without providing any callbacks.

To resolve this problem I start nRF Connect which then causes the scanner to correctly find the friendly name of the devices as well as allowing me to connect to them.

It feels like I am missing something with the scanner, you seem to somehow reset the adapter. How is this done?

Thanks for some great products!

philips77, 2018-12-17:
Are you sure you are finding the correct device?

philips77, 2018-12-18:
Once again nRF Connect saves the day!

PS. There's no magic in the app... I do usual scanning using usual API.

PPS. We have 1+6 at house, how can we reproduce the issue? Are you scanning from UI thread?

nicrgren, 2018-12-18:
Thanks for your responses! I really appreciate it.
Here's a simple gist with the main parts
https://gist.github.com/nicrgren/601299c42b85c101810b1c8109e93cee

Please notice that this issue doesn't always occur. I might notice that I sometime has a headset paired while running this code.

If a bigger example would be helpful please let me know!

philips77, 2019-01-07:
I would suggest removing `.setReportDelay(1000)`. This offloaded batching seems to mess around. 
Also, why did you create the issue in BLE Library, not Android Scanner Compat Library? Should it not be there?