nordicsemi/Android-BLE-Library #73: java.lang.NullPointerException(v2.0.4)

qq1012298028, 2019-01-02:
java.lang.NullPointerException: Attempt to invoke virtual method 'boolean no.nordicsemi.android.ble.ConnectRequest.shouldAutoConnect()' on a null object reference
        at no.nordicsemi.android.ble.BleManager.internalConnect(BleManager.java:797)
        at no.nordicsemi.android.ble.BleManager.access$2100(BleManager.java:113)
        at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.lambda$onConnectionStateChangeSafe$1$BleManager$BleManagerGattCallback(BleManager.java:2450)
        at no.nordicsemi.android.ble.BleManager$BleManagerGattCallback$$Lambda$1.run(Unknown Source:4)
        at android.os.Handler.handleCallback(Handler.java:891)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:207)
        at android.app.ActivityThread.main(ActivityThread.java:7536)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:958)

philips77, 2019-01-07:
Do you know what has happened prior to this crash? Are you using the latest version of the library?

qq1012298028, 2019-01-08:
No,I set it to false.
I think need add Null check to ConnectRequest.

philips77, 2019-01-08:
Adding a null check is easy, but the question is how did it get there.... Do you call `close()` at some point?

qq1012298028, 2019-01-08:
No,I only use disconnect().

philips77, 2019-01-09:
From the stack trace I see that your initial connection has failed and the lib started to reconnect (retry mechanism set). But during the delay, the `mConnectRequest` was nullified.
I will add this null check there in `internalConnect`.

philips77, 2019-01-30:
This should be fixed in 2.0.5 and 2.1.

closed 2019-01-31 by philips77