nordicsemi/Android-BLE-Library #76: WriteCharacteristic returning status 13

kevin-objectstudio, 2019-01-10:
Just want to say this is a great library. I am using both the scanner and BLE Library to refactor an older BLE implementation. I'm running into an issue when writing to a characteristic. I've followed your blinky sample but for some reason when I call writeCharacteristic, I am always seeing a status 13 when onCharacteristicWriteSafe fires inside BleManager.java. I looked up the status, which represents GATT_INVALID_ATTRIBUTE_LENGTH which doesn't quite make sense to me. Are attribute lengths fixed? Is this something that I can change based on my payload length?

Here's my send code:

`public void send(final byte[] data) {
        // Are we connected?
        if (writeCharacteristic == null)
            return;

        writeCharacteristic(writeCharacteristic, data, 0, data.length)
            .with(mWriteCallback).enqueue();
    }`

I call send inside the onDeviceConnected callback. Thanks for your help.

kevin-objectstudio, 2019-01-10:
Okay I believe this has to do with the mtu size, which it appears it's default size is 23 in the BleManager if I'm correct. Do you have any example code in how to write to a characteristic in chunks?

philips77, 2019-01-11:
You may just use the `split()` method:
```java
public void send(final byte[] data) {
    // Are we connected?
    if (writeCharacteristic == null)
        return;

    writeCharacteristic(writeCharacteristic, data, 0, data.length)
        .with(mWriteCallback).split().enqueue();
}
```

philips77, 2019-01-11:
Also, if your device supports Write With Response and Long Write sub-procedure you may set the write type to `WRITE_DEFAULT` (write with response) and Android will automatically use Long Write. But the device needs to prepare a buffer for a longer message, so it's not supported by default. And only works with Write With Response.

kevin-objectstudio, 2019-01-15:
Split method was what I was looking for. Thanks!