nordicsemi/Android-BLE-Library #78: Problem with connection parameters

emclerc, 2019-01-18:
Hello,

I am working on a device with an NRF52. Our android application uses the Android BLE library and we have problems with one specific phone (Huawei P10 lite)  When trying to connect to bootloader.  We can easily reproduce the problem with nrf connect

I suspect the problem is in the phone's bluetooth stack itself, especially in the way it negociates / applies the _connectionInterval_ but am wondering if we may have a workaround with the lib.
Everything happens as if when negotiating the connection interval the phone picks an interval that makes it fail when it applies it.  Is there any mean for the android app to have some control on the connection interval to avoid this situation ?

Following is a detailed description of the problem and the tests i did to get to this conclusion

I have traced exchange between the phone and the device to see what happens and found out it was linked to the connection interval:

- at connection the phone defines an interval of  40 (50 ms)
- the device does a L2CAP connection parameter update request for an interval between 12 (15ms) and 24(30ms)
- the phone accepts it.
- the phone sends a LL_CONNECTION_UPDATE_REQ with interval 17
- at the _instant_ these parameters should apply the phones stops sending connection events
- the device disconnects logically after the negociated timeout

I could tweak our bootloader not to send any connection parameter  update request . In this case the phone behaves like this : 

- at connection the phone defined in interval of 40
- just before doing the service discovery it sends a  LL_CONNECTION_UPDATE_REQ with interval 6
- at end of service discovery it sends a LL_CONNECTION_UPDATE_REQ with interval of 40
- the phone continues to send connection events and everything works fine

If i tweek the device to accept interval between 12 and 40 this is what happens

- at connection the phone defined in interval of 40
- just before doing the service discovery it sends a  LL_CONNECTION_UPDATE_REQ with interval 6
- the device does a L2CAP connection parameter update request for an interval between 12 (15ms) and 40 (30ms)
- the phone accepts it.
- at end of service discovery it sends a LL_CONNECTION_UPDATE_REQ with interval of 17
- at the _instant_ these parameters should apply the phones stops sending connection events , and device disconnect after a timeout

If i tweek the device to accept interval between 40 and  40 this is what happens

- at connection the phone defined in interval of 40
- just before doing the service discovery it sends a  LL_CONNECTION_UPDATE_REQ with interval 6
- the device does a L2CAP connection parameter update request for an interval between 40 and 40 (30ms)
- the phone accepts it.
- at end of service discovery it sends a LL_CONNECTION_UPDATE_REQ with interval of 40
- the phone continues to send connection events and everything works fine

So it really looks like if i give any large interval with L2CAP connection parameter update request the phone picks up interval value of 17 and fails afterwards. If we force a value that we know to work, then it works. 

We will try to find a workaround in the bootloader. But If we could have any mean for the android app to control -even indirectly- the value the phone picks for it's connection interval, this would be far better.

philips77, 2019-01-18:
Hello,

Thank you for investigating the issue before reporting and for the detailed description. It's a nice to know thing for us.
To reply to your question, you are right, it's a matter of the phone, its controller, or the stack, Initial negotiation of connection intervals is beyond what the lib can do. 
Android, since some version, performs service discovery as fast as possible, so it changes the interval to 7.5 ms (interval 6), after which it should change to any value supported by both sides. For Android that is higher than 15 ms. I have no idea why does it like 17 (21.25 ms), and why it can't work with it.

> Is there any mean for the android app to have some control on the connection interval to avoid this situation ?

Yes, you may use `requestConnectionPriority(int)` method in the manager's API. There is mode documentation in code:
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/9ed6541edfd4047336f4f3d552630bbb572bbfc3/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L1805-L1834

emclerc, 2019-01-18:
Thank you for this quick answer.

As I understand, without changing the values our device requests  (15-30ms),  provided (1) the phone's stack reacts correctly to this method, (2) the phone stack works correctly with a setting of 15ms or 30ms, then we might have a workaround for our issue by setting the priority to HIGH or BALANCED. This would be great.

When is the right moment to call requestConnectionPriority ? what is it scope ?
I guess it affects all connections done after the call and be called before establishing the connection. am I right ? 
How is the connection affected if for example the app requests a HIGH priority and at a further time the device sends a L2CAP connection parameter update request with interval in the range of LOW_POWER or not in the range of any of these priorities ?

by the way : for LOW_POWER and BALANCED the comment does not say the same thing than the log "text" strings in the code. there seems to be an inversion here. who is right ? the comment or the log ? 

I am forwarding your answer to the Android developer. He is more familiar than me with the lib and might have further questions.

philips77, 2019-01-18:
> by the way : for LOW_POWER and BALANCED the comment does not say the same thing than the log "text" strings in the code. there seems to be an inversion here. who is right ? the comment or the log ?

Good catch! Fixed internally!

> When is the right moment to call requestConnectionPriority ? what is it scope ?

The priority is valid until any side decides to change connection parameters for any reason. Android doesn't save it, I guess. An unknown situation is when you would like to call it before doing service discovery (the lib doesn't allow for that, but let's say you write it on your own). The discovery changes interval to 7.5 but I have no idea to what value would it be change after that. I would say it stores the range peripheral declared so it would rather select one from the range and forget value from priority, but I did not check it, as it may vary.

Maragues, 2019-01-21:
Hi Phillips, as you well said, there's currently no way for us to change the connection priority before discovering the services

1. The only callback before attempting to discover services is "onDeviceConnected", but invoking `requestConnectionPriority` throws an error. BluetoothGatt is already there, but we can't access it.
2. In order to write our own request priority we need access to `BluetoothGatt` before services have been discovered. Right now, the only point at which we can access it is after services have been discovered.

I understand this is a phone specific issue and that modifying the library solely for it is not good, but I'm out of ideas on what we can do from Android's side.

We can disable OTA for problematic phones, but I'd like to come up with a better solution

philips77, 2019-01-21:
Ideally, I would recommend modifying the firmware part, if it's still possible. You may tweak the connection interval ranges so that it work with all/most phones. 

Indeed the BLE library is constructed in a way that you may execute requests only after service discovery is complete. But, as I said, requesting the priority before service discovery would be pointless, as the interval will change during service discovery and most probably will not be restored afterwards.

closed 2019-01-31 by philips77