nordicsemi/Android-BLE-Library #79: Bluetooth off/on, disconnect not notified

Maragues, 2019-01-23:
I don't receive any callback when bluetooth goes off in the middle of a connection attempt. Is this something we should manage ourselves or is it a bug?

`BleDriver` is the `BleManagerCallbacks` implementation

`BleManager` logs that bluetooth is off, but `BleDriver` doesn't receive any callback after that

```
1st log

08:55:47.216 24225-24225 V/KLNordicBleManager: Connecting...
08:55:47.217 24225-24225 I/BleDriver: C0:4B:24:4B:11:A7 started connecting to the TB
08:55:47.218 24225-24225 D/KLNordicBleManager: gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
08:55:47.218 24225-24225 D/BluetoothGatt: connect() - device: C0:4B:24:4B:11:A7, auto: false
08:55:47.219 24225-24225 D/BluetoothGatt: registerApp()
08:55:47.219 24225-24225 D/BluetoothGatt: registerApp() - UUID=6a5fc436-ce2d-4624-9cd7-81e7c845085f
08:55:47.222 24225-24238 D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
08:55:49.973 24225-24225 D/KLNordicBleManager: [Broadcast] Action received: android.bluetooth.adapter.action.STATE_CHANGED, state changed to TURNING OFF
08:55:49.976 24225-24225 D/KLNordicBleManager: gatt.close()
08:55:49.976 24225-24225 D/BluetoothGatt: close()
08:55:49.976 24225-24225 D/BluetoothGatt: unregisterApp() - mClientIf=5
08:55:50.426 24225-24281 D/BluetoothAdapter: onBluetoothServiceDown: android.bluetooth.IBluetooth$Stub$Proxy@4335981

08:55:58.169 24225-24238 D/BluetoothAdapter: onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@25e5f1f
//No reconnect attempt
```

```
2nd log

09:06:17.395 25147-25147 E/BleDriver: onLinkLossOccurred: Ble Manager lost connection with C0:4B:24:4B:11:A7.
09:06:17.401 25147-25147 I/BleDriver: C0:4B:24:4B:11:A7 started connecting to the TB
09:06:17.403 25147-25147 D/KLNordicBleManager: gatt.connect()
09:06:17.933 25147-25147 D/KLNordicBleManager: [Broadcast] Action received: android.bluetooth.adapter.action.STATE_CHANGED, state changed to TURNING OFF
09:06:17.937 25147-25147 D/KLNordicBleManager: gatt.close()
09:06:17.937 25147-25147 D/BluetoothGatt: close()
09:06:17.937 25147-25147 D/BluetoothGatt: unregisterApp() - mClientIf=7
09:06:18.308 25147-25159 D/BluetoothAdapter: onBluetoothServiceDown: android.bluetooth.IBluetooth$Stub$Proxy@67f63cc
09:06:25.133 25147-25184 D/BluetoothAdapter: onBluetoothServiceUp: android.bluetooth.IBluetooth$Stub$Proxy@6dd615
//No reconnect attempt
```

Tested on Nexus 6P, Android 8.1.0

Thanks!

Maragues, 2019-01-23:
We haven't migrated yet, it's in a near future scope.

So, I'd a appreciate the 2.0.5 release that also includes the `filter`

Thanks for the quick reply

Maragues, 2019-01-29:
Hi @philips77 sorry to bother you, but do you have an estimated release date for 2.0.5 that includes this bugfix (if it's a bug)?

Thanks!

philips77, 2019-01-29:
I had to give my normal daily computer to IT for some time, I'll look into it when I get it back. 2.0.5 should be out by the end of week.

philips77, 2019-02-01:
2.0.5 has been released. Without the `filter` as this was only a bug fixing release.
Filter will be added in version 2.1, that will actually be migrated to AndroidX. Time to migrate! :)

PR #82 (2019-02-01): #79 fixed
This PR should fix #79. The Bluetooth state event was ignored when the device was not connected.
With this changes it will signal all waiting requests even if connection was not established.
closed 2019-02-01 by philips77