nordicsemi/Android-BLE-Library #80: Memory leaks? Strange behavior.

barkov00, 2019-01-24:
Hi!
I use BleManager inside the Service. 
The following sequence of actions leads to the complete inoperability of the library:
1. Start service, call connect().enqueue()
2. Device was connected
3. Call bleManager.close(), stop the service
4. Restart service after 200...400 ms, create new BleManager instance or use singlton instance from Application class.
5.  Bluetooth device still reconnected to smartphone, but callbacks from BleManager not called.

Maybe I incorrectly finish work with BleManager?

philips77, 2019-01-25:
Hello,
Calling `close()` is not required. It's enough to call disconnect. It will close the manager automatically.
Also, disconnecting doesn't mean that the device will disconnect. It may be still connected by another BluetoothGatt client in your app or another (like nRF Connect).

philips77, 2019-01-31:
HI, did you manage to solve the issue?

barkov00, 2019-01-31:
Hello! Not.

I tried the following methods:
1. Call close before terminating the service.
2. Store an instance of NRFManager in a singleton inside the Application class
3. Call disconnect and terminate the service after the callback has completed.

The problem always appears if you restart the service immediately after its completion and call the connect method. 
If after the completion of the service, wait 5 seconds - then the problem does not appear.
I think this is related to memory management in java.

svvashishtha, 2019-03-13:
@philips77 I also verified this issue. 
Steps to reproduce:

1. Exit the app while the device is **connecting**. (The service that connects to ble device stops it when all components unbind from it i.e. when app is closed)
2. Reopen the app within 5 seconds
3. Try to connect with the device again. 
The device will not connect and connection fail callback is also not called. 

Similar condition
1. Connect with a device. 
2. Call disconnect on the device
3. Connect with another device.
The device will not connect and connection fail callback is also not called.

svvashishtha, 2019-03-14:
@philips77 I found a solution for the above problem.
The solution is to enqueue a disconnect request even if the device is not connected but is connecting. 
And use a singleton instance of BleManager so that multiple connections are not opened at the same time. 
This way the connect request will be executed only after the previous disconnect request(if any) have been executed.

barkov00, 2019-03-14:
> @philips77 I found a solution for the above problem.
> And use a singleton instance of BleManager so that multiple connections are not opened at the same time.
Where to store this instance?
Does this method work when you stop the service?

svvashishtha, 2019-03-14:
I use dagger to inject BleManager in my service. 
When I stop the service I enqueue a disconnect call. 
Reconnecting with same device or other works as expected when I reenter into application
(The service that maintains the connection stops when you exit the application and is started whenever you enter).

barkov00, 2019-04-03:
@svvashishtha Is it same like to store BleManager instance in the Application class?

svvashishtha, 2019-04-04:
@barkov00 Yes. Basically, you need to avoid instantiating BleManager more than once, if you are connecting to a single device. 
If you have to close the connection and then reopen it, then using the same instance makes sure that disconnect and connect requests are executed sequentially while that may not be the case if you reinstantiate BleManager for new connection because the previous connection might not have been closed yet.

barkov00, 2019-04-05:
@svvashishtha None of the suggested solutions helped me. I don’t understand how people use this library, even if it isn’t possible to connect to the device in most cases.

philips77, 2019-04-05:
Hi @barkov00, do you have your code on GitHub, so that I could take a look?
What I would do is:

    Start service, call connect().enqueue()
    Device was connected
    Call disconnect().enqueue(), wait for done() and stop the service
    Restart service after 200...400 ms, create new BleManager instance

Did you try like this? The `close()` method is called automatically when the device disconnected.

philips77, 2020-06-30:
Closing the issue due to inactivity.

closed 2020-06-30 by philips77