nordicsemi/Android-BLE-Library #81: E/bt_btif: bta_gattc_process_indicate, ignore HID ind/notificiation

barkov00, 2019-01-26:
Hello!
After restarting application and reconnecting to device i have folowing errors in LogCat:

```
2019-01-26 10:06:22.109 24296-24340/? E/bt_btif: bta_gattc_process_indicate, ignore HID ind/notificiation
2019-01-26 10:06:24.108 24296-24340/? E/bt_btif: bta_gattc_process_indicate, ignore HID ind/notificiation
2019-01-26 10:06:26.106 24296-24340/? E/bt_btif: bta_gattc_process_indicate, ignore HID ind/notificiation
2019-01-26 10:06:28.105 24296-24340/? E/bt_btif: bta_gattc_process_indicate, ignore HID ind/notificiation
2019-01-26 10:06:30.104 24296-24340/? E/bt_btif: bta_gattc_process_indicate, ignore HID ind/notificiation
2019-01-26 10:06:32.102 24296-24340/? E/bt_btif: bta_gattc_process_indicate, ignore HID ind/notificiation
2019-01-26 10:06:34.101 24296-24340/? E/bt_btif: bta_gattc_process_indicate, ignore HID ind/notificiation
```

The period of appearance of these messages is equal to the period of sending notifications from the Bluetooth device, but onNotificationReceived is not called.
Are there any ways to solve this problem?

Lenovo K6 Power, Android 7

philips77, 2019-01-26:
Are you trying to receive a notification from HID service? Those, since Lollipop, are protected and available only for system apps. Otherwise you'd be able to listen to passwords, etc. There is no workaround.

closed 2019-02-20 by philips77

anwarhamr, 2021-12-05:
I'm getting this message as well however my notifications are making it to the app.  The messages are flooding logcat and with such frequency are probably affecting performance/cpu.  Anyone able to fix?