nordicsemi/Android-BLE-Library #84: Issues importing library into Xamarin Forms C#

sdinnes, 2019-02-04:
I am attempting to use this library in a Xamarin Forms C# project; you can bind a Java library to an Android .Net assembly using a special project type in Visual Studio.  This process usually involves doing some tweaking to the bindings using a project xml file included in the VS project.

I have managed to fix several binding issues but have come up against one that has so far defeated me: in the error list, I get many errors of the type

The type 'ConnectRequest' already contains a definition for 'cb_timeout_J'

My analysis has led me to believe that it is a result of class TimeoutableRequest (which is abstract) and descendant class ConnectRequest both having method "timeout".  Although declared override in ConnectRequest, it seems the bindings are mapping it twice, leading to the above errors.

I have  tried renaming the method and the parameter names, to no avail.

Do you have any tips for getting this working?

philips77, 2019-02-05:
Hmm... I see the `@IntRange(from = 0)` was missing in `timeout(long)` in `ConnectRequest` and `TimeoutableValueRequest` classes, did you try adding it? Perhaps the signature wasn't exact?
That is the only thing that differs `timeout` from any other method that also is overridden in `ConnectRequest`.

sdinnes, 2019-02-05:
Thanks for the feedback!  Late last night I found what I think is the solution; I'm working on testing it out now in a test app.

I had noticed that TimeoutableRequest had no visibility qualifier, whereas Request and ConnectRequest are both public.  I tried fixing this in the metadata.xml to no avail, so finally I did what I'd wanted to avoid, and changed the java - only adding the public qualifier to TimeoutableRequest.  This immediately fixed all of the binding errors!

I think the clue was in the generated C#, it showed ConnectRequest as being directly descended from Request, omitting TimeoutableRequest entirely.  (In C# there are strict rules about descendant visibility).

I'll post an update later today hopefully on whether this fixed the issue.

philips77, 2019-02-05:
I could make the middle classes public if it helps. Please share your results.

sdinnes, 2019-02-19:
In the end I only had to do that one change, all else was just getting to grips with general Java binding issues.  The new library works massively better than any I used previously!

In time, I may put an example project up on GitHub and, subject to management approval in my company, may attempt to create a NuGet package for this for use in Visual Studio.

closed 2019-02-19 by sdinnes

lfongaroScp, 2020-01-07:
@sdinnes Did you manage to bind this library with Xamarin?
Is your code public available?

Thank you

sdinnes, 2020-01-08:
@lfongaroScp - thanks for the reminder!  I have a sample Xamarin Forms project which includes the bound libraries (scanner and connect) and demonstrates how to use them with some simple iBeacon BLE commands.  As this derives from a commercial product I work on for my employer, I need to obtain their approval before I make it public.  Watch this space!

sdinnes, 2020-01-15:
@lfongaroScp - I have made my repo public:

[https://github.com/sdinnes/NordicAndroidBle](url)

geervani, 2020-05-26:
@sdinnes [this](https://github.com/sdinnes/NordicAndroidBle) link is broken.

philips77, 2020-05-26:
I fixed the link.