nordicsemi/Android-BLE-Library #85: BluetoothServiceJni: android.os.DeadObjectException: Transaction failed on small parcel; remote process probably died

StanislasGaitUp, 2019-02-06:
Hi,

I am using your library (v2.1.0) to communicate with Nordic devices through notifications. The phone is the central in my setup and every device notifies data on a characteristic as fast as he can. Right now, the app does not process data at all, it just logs that it received something from device X.
Unfortunately, when using a cheap phone (Asus ZenFone 3 Max) and multiple peripheral, I start to see multiple errors in logcat and my app stops responding.

Here is a sample of the error I am getting:

```
02-06 09:14:45.027 3370-3401/? E/JavaBinder: !!! FAILED BINDER TRANSACTION !!!  (parcel size = 208)
02-06 09:14:45.028 3370-3401/? E/BluetoothServiceJni: An exception was thrown by callback 'btgattc_notify_cb'.
02-06 09:14:45.028 3370-3401/? E/BluetoothServiceJni: android.os.DeadObjectException: Transaction failed on small parcel; remote process probably died
        at android.os.BinderProxy.transactNative(Native Method)
        at android.os.BinderProxy.transact(Binder.java:511)
        at android.bluetooth.IBluetoothGattCallback$Stub$Proxy.onNotify(IBluetoothGattCallback.java:848)
        at com.android.bluetooth.gatt.GattService.onNotify(GattService.java:920)
```

I looked on stack overflow and it seems to come from the fact that there is no available space left in the binder buffer https://stackoverflow.com/questions/45432647/android-throw-deadobjectexception-with-log-transaction-failed-on-small-parcel
Is there a way to detect this problem and to reduce the amount of packet accepted by the BLE in order to empty this buffer ?

philips77, 2019-02-06:
I don't think it's possible from app/lib. You'd need to wait on the remote side, or require a write request from the phone from time to time to make sure both sides are on the same page.

StanislasGaitUp, 2019-02-06:
Thank you very much for your suggestion, we will try to find a workaround this problem

closed 2019-02-06 by StanislasGaitUp

apoorvkarkare, 2021-01-24:
@StanislasGaitUp Hi there, I have very less experience in BLE app development. Though I am not using this library, but I am facing exactly same issue here. Can you help me out telling that how exactly did you handle it?