nordicsemi/Android-BLE-Library #88: NullPointerException on service discovery

vicobz, 2019-02-27:
Hey!

Just collected that crash from Firebase Crashlytics, here is the stacktrace:

```Fatal Exception: java.lang.NullPointerException
Attempt to invoke virtual method 'android.bluetooth.BluetoothDevice android.bluetooth.BluetoothGatt.getDevice()' on a null object reference
no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.onDeviceReady (BleManager.java:2207)
no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest (BleManager.java:3146)
no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest (BleManager.java:3210)
no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest (BleManager.java:3210)
no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.nextRequest (BleManager.java:3210)
no.nordicsemi.android.ble.BleManager$BleManagerGattCallback.onServicesDiscoveredSafe (BleManager.java:2601)
no.nordicsemi.android.ble.MainThreadBluetoothGattCallback.lambda$onServicesDiscovered$1 (MainThreadBluetoothGattCallback.java:112)
no.nordicsemi.android.ble.-$$Lambda$MainThreadBluetoothGattCallback$js3Daz29T8D5sg6HeYmRg0i-uuU.run (Unknown Source:6)
```

philips77, 2019-03-14:
I think the proper solution to this fix would be to hide the `close()` method in `BleManager`. It's automatically called by the library when the device gets disconnected and there is no need to call it manually.
In your case, the device gets connected and starts service discovery, and it looks like `close()` was called at this moment which cleared `mBluetoothGatt` and `mBluetoothDevice` references, as all `nextRequest` methods are called in the else section.
https://github.com/NordicSemiconductor/Android-BLE-Library/blob/12441cb4fe0074c3e2267500bd28621d288cb398/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L3203-L3212
This goes through all Requests from the init queue and finally calls `onDeviceReady()`, where it fails on NPE.

I will add some checks here and there, but the solution is not to call `close()`.

vicobz, 2019-03-14:
@philips77 Thanks for your reply, we effectively call `close()` manually to ensure a proper configuration resetting before trying to reconnect.

We'll clean this behavior asap from our side.

Is your version 2.1.1 milestone actually planned for today ?
cc @pvarry

philips77, 2019-03-15:
2.1.1. was released yesterday, but after all I didn't implement anything related to this issue. I need more time to investigate how that could occur, and I have no time right now. Are you able to replicate the issue? Does not calling `close()` help?

vicobz, 2019-03-15:
Actually not, I only observed the issue once from Firebase Crashlytics

philips77, 2020-03-11:
Hello, did you tried the new version, either 2.1.1 or 2.2.0-alpha? Can we close this issue?

philips77, 2020-06-30:
This should be fixed in 2.2.1, or before.

closed 2020-06-30 by philips77