nordicsemi/Android-BLE-Library #91: How to get access to discovered services

michal-narajowski, 2019-03-07:
Hi,

I can see that the service discovery starts automatically after the connection, but I can't find a way to access these discovered services. I tried looking through the code in BleManager class and I don't think the library exposes BluetoothGatt object or has an API to access discovered services.

I would appreciate your help.

philips77, 2019-03-07:
Have a look at this method in nRF Blinky:

philips77, 2019-03-07:
There is also `isOptionalServiceSupported(BluetoothGatt)` method.

michal-narajowski, 2019-03-08:
@philips77 Thanks for your help. I think I can use that. 

Is there a way to use this library from a peripheral side? E.g. I start advertising and then after the connection I use the same API as when I'm central.

philips77, 2019-03-09:
You can use it as a peripheral, but will only work for one side of the connection, the client side. When the target device connects to you, and you'll get the server callback, just use the BLE Library with that BluetoothDevice that you got as a parameter. It will discover services in the other side and you'll be able to use it normally. 
But the server side is missing. There's no option to define gatt server services, advertise or do server operations, like send notifications/indications, wait for write/read, wait for mtu or phy request, etc. That's in fact missing and should be added one day. In nRF Toolbox, in Proximity profile, we do this separately.

closed 2019-03-14 by philips77