nordicsemi/Android-BLE-Library #92: Disconnection after timeout causes BT stack crash and subsequent DeadObjectException

wcauchois, 2019-03-07:
Hi,

I've been testing testing my BLE app on an LG phone, the LG G6 (model number: LG-H872, android version: 8.0.0). I'm connecting to a device using the following parameters:

```kotlin
                connect(device)
                        .retry(2, 1500)
                        .useAutoConnect(false)
                        .timeout(30 * 1000L)
                        .await()
```

I'm observing that some times the connection attempt times out at which point this BLE library closes the connection before raising the exception to my code (`Cancelling connection...`). After that I get the expected InterruptionException.

However, moments later the entire system bluetooth stack crashes and restarts itself. I think the relevant log message is `bt_hci: command_timed_out: Waited 2003 ms for a response to opcode: 0x200e *matches timer*` -- an HCI command times out and it restarts the bluetooth process.

After **that**, there's a DeadObjectException emanating from BleManager.close, presumably as it calls close() on some no-longer-relevant BluetoothGatt object.

Here are some system logs illustrating the entire flow, starting from connection thru crash through DeadObject:

https://gist.githubusercontent.com/wcauchois/6054af38f156623e7288704bf0b8d120/raw/d1c0c3c1b17b8ef4b5e3ca8d5da8179e046698be/connection_log.txt

Can anyone help me with this issue? I'm not sure if it's an issue with this library or the way I'm using it or simply with the phone. It's OK if a connection times out, but for it to crash the bluetooth stack and display a "Bluetooth has stopped" message to the user is definitely ideal.

Thanks in advance.

philips77, 2019-03-14:
Hello,
Well.. crashing the Bluetooth stack is definitely Android or phone issue.
Could you set the timeout to < 30 seconds? At 30 seconds there is a system timeout and with yours set as well to 30 they may come simultaneously. I would recommend around 20-25. Check if it helps. If user is waiting, I would say even 20 is a long time. The timeout should be 2-3 times longer than the average waiting time in case of success. There is no need to wait longer. If it didn't in first 3 seconds, it will not after that as well.

When both timeouts occur near the same time, the app may be cancelling an already closed connection and this is when things get dirty, I think.

philfi, 2019-03-18:
Hi @philips77, thanks for the response! I'm a coworker of @wcauchois and have been digging into this a bit more. Setting the timeout less than 30 seconds definitely helped reduced the frequency of the crashes, thanks for that suggestion! 

Unfortunately the crashes are still occurring though it seems (based off some anecdotal evidence found on various forums) like there may be a specific issue with a T-Mobile security update for the LG G6 that we've been testing with. 

That being said if you have any other suggestions on how to ensure stability of the BLE stack on Android or how to better troubleshoot when issues do occur, we would love to hear them!

philips77, 2019-03-19:
No, I don't know more tips. I tried to include all I know in the library so you shouldn't have to worry about them, but as you see some phones needs special attention. If you discover anything that helps, feel free to submit a PR. Also, from my experience, many things can be improved on the peripheral side. By changing connection interval, improving interoperability, etc.

closed 2019-04-09 by wcauchois