nordicsemi/Android-BLE-Library #95: Bluetooth service crash (multi device connection)

mcolunas, 2019-03-18:
Hello, we are facing an issue by connecting several sensors (UART nrf52832) to an Android Application ( Android Nougat 7.0) and we are facing disconnections (this happens randomly) with the sensors by the bluetooth service crashes and goes down, killing the connection.
We have tested several approaches (removing gatt operations (write and reads) and only receive notifications), however, the issue continues to happen.

Here it goes some logs of the issue:

3-14 11:17:27.625 3226-3759/? E/bt_vendor: can_perform_action: BT turn off twice before BT On(ref_count=0)
2019-03-14 11:17:27.627 3226-3759/? E/bt_vendor: Stopping HCI filter as part of CTRL:OFF
2019-03-14 11:17:27.627 3226-3759/? I/bt_vendor: stop_hci_filter: hci_filter has been stopped already
2019-03-14 11:17:27.627 3226-3759/? I/bt_vendor: bt-vendor : get_bt_soc_type
2019-03-14 11:17:27.627 3226-3759/? I/bt_vendor: qcom.bluetooth.soc set to rome
2019-03-14 11:17:27.629 3226-3759/? I/bt_vendor: Write 0 to rfkill
2019-03-14 11:17:27.631 3226-3319/? E/BluetoothDataManager: getUartErrorInfo NONE
2019-03-14 11:17:27.631 3226-3319/? E/BluetoothDataManager: "LO_MFN":"29","LO_LMP":"8","LO_SUB":"602","LO_FWV":"Rome Ver. 0x131","ERROR_REASON":"3","UART":"NONE"
2019-03-14 11:17:27.631 3226-3319/? E/BluetoothDataManager: Failed to call QmBigDataModule API

02-26 10:50:06.254 3204-3293/? D/BluetoothAdapterService: getAdapterService() - returning com.android.bluetooth.btservice.AdapterService@57a0d3c
02-26 10:50:06.266 3204-3293/? D/BluetoothDataManager: Call QmBigDataModule API additionally
02-26 10:50:06.267 3204-3293/? E/BluetoothDataManager: getUartErrorInfo NONE
02-26 10:50:06.267 3204-3293/? E/BluetoothDataManager: “LO_MFN”:“29",“LO_LMP”:“8",“LO_SUB”:“602",“LO_FWV”:“Rome Ver. 0x149",“ERROR_REASON”:“3",“UART”:“NONE”
02-26 10:50:06.267 3204-3293/? E/BluetoothDataManager: Failed to call QmBigDataModule API
02-26 10:50:06.279 3204-3596/? I/bt_vendor: stop_hci_filter: hci_filter has been stopped already
bt-vendor : get_bt_soc_type
qcom.bluetooth.soc set to rome
02-26 10:50:06.382 3204-3596/? I/bt_vendor: stop_hci_filter: hci_filter has been stopped already
bt-vendor : get_bt_soc_type
qcom.bluetooth.soc set to rome
02-26 10:50:06.427 3247-3247/? D/A2dpProfile: Bluetooth service disconnected
02-26 10:50:06.428 3247-3247/? D/BluetoothSap: Proxy object disconnected
02-26 10:50:06.428 2830-2894/? E/BluetoothManagerService: MESSAGE_BLUETOOTH_SERVICE_DISCONNECTED: 1
02-26 10:50:06.428 2830-3127/? I/ActivityManager: Process com.android.bluetooth (pid 3204) has died(50,802)
02-26 10:50:06.429 3247-3247/? D/SapProfile: Bluetooth service disconnected
02-26 10:50:06.429 2830-3127/? W/ActivityManager: Scheduling restart of crashed service com.android.bluetooth/.btservice.AdapterService in 1000ms
02-26 10:50:06.429 3247-3247/? D/BluetoothPan: BluetoothPAN Proxy object disconnected
02-26 10:50:06.429 3247-3247/? D/PanProfile: Bluetooth service disconnected
02-26 10:50:06.429 2830-3127/? W/ActivityManager: Scheduling restart of crashed service com.android.bluetooth/.map.BluetoothMapService in 11000ms
02-26 10:50:06.429 3247-3247/? D/BluetoothMap: Proxy object disconnected
02-26 10:50:06.429 3247-3247/? D/MapProfile: Bluetooth service disconnected
02-26 10:50:06.430 3247-3247/? D/BluetoothPbap: Proxy object disconnected
02-26 10:50:06.430 3247-3247/? D/PbapServerProfile: Bluetooth service disconnected
02-26 10:50:06.430 3247-3247/? D/BluetoothInputDevice: Proxy object disconnected
02-26 10:50:06.430 3247-3247/? D/HidProfile: Bluetooth service disconnected
02-26 10:50:06.430 3247-3247/? D/HeadsetProfile: Bluetooth service disconnect

philips77, 2019-03-19:
Hi, this looks like the phone issue. Did you try other phones? I don't know if anything can be done about it...

mcolunas, 2019-03-19:
The strange thing is that connected with 4/5 trackers the issue does not happen. With less than 3 devices connected it happens. We also have verified that without writing characteristics, it is less prone to the crash, so it seems to be related with this.

mcolunas, 2019-03-19:
I send more details about the communication with the devices if it helps:

We read characteristic values each 60 seconds.
Write characteristics for each sensor (each 3 seconds).
Have characteristic notifications each 20 miliseconds.

closed 2019-04-02 by mcolunas