nordicsemi/Android-BLE-Library #99: Samsung phones time out after 5 seconds

weliem, 2019-04-08:
Hi,

I noticed you now check if an error 133 while connecting is a timeout or not. You use CONNECTION_TIMEOUT_THRESHOLD set to 20 seconds. I did some testing on my Samsung phones (S7, A8, S5) and it looks like Samsung phones time out after exactly 5 seconds...but indeed all other phones I have time out after 30 seconds.

Perhaps 20 seconds is not the right number for all phones?

philips77, 2019-04-09:
Interesting... I will look into it more closely. Thank you for reporting it. Unfortunately, this will have to wait, as I'm buried with other tasks.

junebeans, 2022-03-14:
Was there any update on the issue? I wonder if there's any workaround for this. I'm observing the same issue in several Samsung Android phones and tablets. So far what I have on the list are Galaxy S21 and Galaxy Tab Active Pro.

philips77, 2022-03-15:
No, not yet, sorry. Thank you for providing the list of impacted devices.
May I ask for some help with debugging? Could you install nRF Connect app for Android, scan for your device until it shows up on the scanner view, turn off your device so that connection attempt would fail, and click CONNECT button? Then, could you paste here the logs you get on the device tab. Thank you!

philips77, 2022-09-30:
Closing a very long issue. To set a connection timeout use `.timeout(..)` method in the `ConnectRequest`.

closed 2022-09-30 by philips77

junebeans, 2022-09-30:
Hi @philips77! I noticed the comment just now and wanted to let you know the software mentioned above was using the timeout of 30 seconds. However for Samsung devices particularly it was 5 seconds. Yet as it is closed, once I get a chance to update the library version and test it on a Samsung device, I'll let you know if it is still an issue or not.

philips77, 2022-10-04:
Since the latest alpha version the connection attempt will also cancel when started from a coroutine, for example:
```kotlin
suspend fun connect() = withTimeout(TIMEOUT) {
   connect(device)
      .useAutoConnect(false)
      .retry(3, 100)
      .suspend()
}
```

At least I hope :)