nordicsemi/Android-BLE-Library #101: Unable to run app in Android Studio until "Instant Run" disabled

WhatDoYouKnow, 2019-04-22:
Hi
Just trying to run an app for BLE and the Blinky app kept stopping without getting started.
Found another github project which suggested disabling "Instant Run" (File > Settings > Build, Execution, Deployment > Instant Run).
Doing this allowed nRF Blinky to run in my environment. Add this to the front page?
Thanks

WhatDoYouKnow, 2019-04-22:
Error message starts with: Didn't find class "androidx.lifecycle.ProcessLifecycleOwnerInitializer" on path:
Frustrating until this simple change fixed it.

philips77, 2019-04-23:
Thanks, I'll add it to the Readme.

JohnIDPal, 2019-07-31:
Hi, unable to run this app in android studio. Can you please set the missing files in the mainfest

philips77, 2019-07-31:
This is not an app, it's a library. Please, check out https://github.com/NordicSemiconductor/Android-nRF-Blinky for how to use it in an app.

philips77, 2020-06-30:
Instant Run is no longer a thing. Closing the issue.

closed 2020-06-30 by philips77