nordicsemi/Android-BLE-Library #102: call write method

gaganAriel, 2019-04-29:
Please let me know how to call write method. i have no experience

philips77, 2019-04-29:
Hello,
Have a look at our examples. For instance, in nRF Blinky app, here's how the method is used: [nRF Blinky sending ON/OFF commands](https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/065ab072219f0c9a7fed1e00661effe6d75605eb/app/src/main/java/no/nordicsemi/android/blinky/profile/BlinkyManager.java#L192)
A different example you will find here: [nRF Toolbox -> UART profile](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/08b67a28c9ab5c0dbc49be619c27209e2f955e30/app/src/main/java/no/nordicsemi/android/nrftoolbox/uart/UARTManager.java#L131) or here: [nRF Toolbox -> Proximity profile](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/08b67a28c9ab5c0dbc49be619c27209e2f955e30/app/src/main/java/no/nordicsemi/android/nrftoolbox/proximity/ProximityManager.java#L127).

philips77, 2019-04-29:
Also, for future questions like that, please use [Nordic DevZone](https://devzone.nordicsemi.com/) or [StackOverflow with tag `android ble library`](https://stackoverflow.com/questions/tagged/android-ble-library).

closed 2019-04-29 by philips77