nordicsemi/Android-BLE-Library #104: Does not close() GATT client if the connection attempt times out

wcauchois, 2019-05-03:
I'm connecting with the following parameters:

```
connect(device)
        .retry(2, 1500)
        .useAutoConnect(false)
        .timeout(CONNECT_TIMEOUT_MS)
        .await()
```

Sometimes my connection attempts fail with the FailCallback code of -5, request timed out. This is to be expected. However, I believe that after the connection fails and the BleManager disconnects, it does not close() its BluetoothGatt instance. Perhaps its meant that the user of the library should call BleManager.close()? However, if the user of the library neglects to do that, I believe it can result in BluetoothGatt clients "leaking". In my case this eventually leads to the dreaded "MAX GATT clients reached: 32" error.

Here are some example logs from a connection attempt:

```
FiManager	Connecting to device FF:0B:44:11:F2:3D
FiManager	Connecting...
FiManager	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, LE 1M)
FiManagerCallbacks	onDeviceConnecting
FiManager	Cancelling connection...
FiManagerCallbacks	onDeviceDisconnecting
FiManager	gatt.disconnect()
FiManager	Disconnected
FiManagerCallbacks	onDeviceDisconnected
Error connecting to (FF:0B:44:11:F2:3D)
java.lang.InterruptedException
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:742)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.access$runSafely(CoroutineScheduler.kt:60)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:594)
	at kotlinx.coroutines.DispatchedTask.run(Dispatched.kt:233)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:32)
	at com.barkinglabs.fi.ble.FiManager$connectAndInterrogate$3.invokeSuspend(FiManager.kt:318)
	at no.nordicsemi.android.ble.TimeoutableRequest.await(TimeoutableRequest.java:128)
```

Interestingly if you attempt to reuse the same BleManager to connect to the peripheral again, [this code](https://github.com/NordicSemiconductor/Android-BLE-Library/blob/master/ble/src/main/java/no/nordicsemi/android/ble/BleManager.java#L772) will close() the BluetoothGatt before reattempting connection. Additionally, if after being connected to a device you explicitly call `BleManager.disconnect()`, this also closes the BluetoothGatt.

However, I think it might also be necessary to close() the BluetoothGatt on a connection timeout. Does this align with anyone else's experiences? Is it meant for the user of the library to try/catch and then close() manually after a connection exception?

philips77, 2019-05-04:
Sure the `close` should be called automatically in that case. I'll look at it, but I'm busy with other tasks now. It may be trivial, I'll try to fix it asap.

wcauchois, 2019-05-08:
Thank you @philips77! I would also be willing to try to implement this, if you could point me to a place in the code that would be best to put the gatt close().

wcauchois, 2019-05-08:
Actually, does this seem like it would work? https://github.com/barkinglabs/Android-BLE-Library/commit/785c5fad43e63f8005333124f2328c7e2251cce3. I'm testing it right now with real devices, I think it's working. I'm not sure if that's a clean/safe place to put it though.

philips77, 2022-03-21:
I'm closing the issue due to inactivity.

closed 2022-03-21 by philips77