nordicsemi/Android-BLE-Library #105: onClientConnectionState() - status=22 clientIf=5

gaoyouhuang, 2019-05-07:
hi~
[image]

Now I pair my mobile phone with the device and connect it with Bluetooth. Then I delete the matching information of the device. Then the mobile phone connects the device through Bluetooth, and it finds that the device can not connect, and returns to state 22. What does state 22 mean?

gaoyouhuang, 2019-05-07:
Now what I do is, when I encounter the state 22, remove Bond and connect Gatt, but I don't think that's a good way to do that.

philips77, 2019-05-07:
Error 22 (0x16) is: https://github.com/NordicSemiconductor/Android-BLE-Library/blob/5e0e2f08c309a6de2376d9b8705c83f9e9a80d56/ble/src/main/java/no/nordicsemi/android/ble/error/GattError.java#L38
If you want to remove bond information you have to do this on the phone and on the device together. Otherwise one side will try to re-initiate encryption and will fail. Android doesn't remove bond information automatically when this happens. User has to do this in Bluetooth settings, or using this hidden method to remove bonding. Also, error 22 may happen for some other reasons than the one described.

closed 2019-05-07 by philips77

gaoyouhuang, 2019-05-07:
thank you very much

georrge1994, 2021-11-05:
@philips77 you are the big professional in BT. Can you advice me? I get timeout disconnection for bond device after 30 seconds. If I clear cache on client device, error will gone and it works until I restart me gatt server. I can not refresh cache for all clients, I don't have access to some of clients. How I can find and resolve the real reason of local host side? Anyway thank you for your lib